cocotb/cocotb #2727: Does cocotb support the vivado emulator?

qgzln, 2021-09-30:
Does cocotb support the vivado emulator?Do you plan to support the vivado emulator ?

ktbarrett, 2021-09-30:
cocotb does not support the Vivado simulator and cannot because Vivado does not provide the VPI, VHPI, FLI, or similar procedural interface.

closed 2021-09-30 by ktbarrett

MrDrGrumpy, 2022-03-05:
> cocotb does not support the Vivado simulator and cannot because Vivado does not provide the VPI, VHPI, FLI, or similar procedural interface.

Would the Xilinx Simulator Interface (XSI) provide a similar alternative to the VPI?

ktbarrett, 2022-03-05:
> Would the Xilinx Simulator Interface (XSI) provide a similar alternative to the VPI?

No, it's missing a serious amount of functionality necessary for cocotb to work.

bp-ouhaha, 2023-12-07:
Nowadays, there is DPI: https://docs.xilinx.com/r/en-US/ug900-vivado-logic-simulation/Direct-Programming-Interface-DPI-in-Vivado-Simulator

Would that be enough?

marlonjames, 2023-12-07:
No, cocotb relies on VPI callbacks.

themperek, 2024-01-23:
You can try https://github.com/themperek/cocotb-vivado. It is early version and  very limited because of limited XSI interface.