cocotb/cocotb #2762: TestFactory: Passing individual *args/**kwargs/ subcase identifier to run_test has changed/ stopped working in 1.6.0

c-93, 2021-11-03:
Using TestFactory, I am facing difficulties in providing test-individual arguments to run_test() in cocotb==1.6.0.

I will elaborate on an example using the test-inidividual `subcase` string:  

1. Prior usage ( with cocotb <= 1.5.2):

1.1 Assume a run_test, which gets a per-test individual string e.g. `subcase` as an argument.
```
[...]
@cocotb.test(timeout_time=200000, timeout_unit="us")
def run_test(dut, subcase=None):
	if not subcase:
		print("ERROR: Subcase is none!")
        [...]
```
1.2 Assume two different example test factories, which could provide the `subcase` string with cocotb <=1.5.2:

1.2.1
```
factory = TestFactory(run_test)
factory.add_option("arbitraryName",  longListOfSubcases)
factory.generate_tests()
```
1.2.2
```
factory = TestFactory(run_test)
 for case in longListOfSubcases:
 	setattr(mod, arbitraryVaryingName, _create_test(run_test, arbitraryVaryingName, "nodoc", mod, case))
```

2. Current problem:
I failed when trying different approaches for providing test-inidividual arguments to run_test(). Either with implicit arguments like substring or explicit arguments it did not work for me. I am not sure, whether it is a bug, or me using the software wrongly.

Suggestion:
Please verify how to provide test-inidividual arguments in cocotb==1.6.0. Thank you.

ktbarrett, 2021-11-03:
Not sure what you mean by "test-individual arguments".

> ```
> factory = TestFactory(run_test)
> factory.add_option("subcase",  longListOfSubcases)
> factory.generate_tests()
> ```

This is the only way the `TestFactory` was meant to be used. Is that not working? Can you provide a minimal failing example and the error message?

c-93, 2021-11-03:
Thanks for your reply.  
Maybe, "test-individual arguments" was not the accurate term.  
With "test-individual arguments", i mean a string, which is different per subcase. I am using the `subcase string`, to identify and then load a subcase-individual binary into a RISC-V processor.

In `cocotb 1.5.2`, I can use the prior-mentioned code to setup a TestFactory:
```
factory = TestFactory(run_test)
factory.add_option("subcase",  longListOfSubcases)
factory.generate_tests()
```

I can then run the following test:
```
def run_test(dut, subcase=None):`
    print(subcase) # will work with cocotb==1.5.2, will fail with cocotb==1.6.0
```

In `cocotb==1.5.2`, and as expected, the function-body of `run_test` has access to the `subcase` string. `subcase` is not `None`, but corresponds to the individual `subcase`. More precisely, `subcase` is an individual element out of `longListOfSubcases`.

With `cocotb==1.6.0`, `run_test` cannot access `subcase` anymore. Without error `subcase` will always default to `None`. If no default is given, it will cause an error.

So the question is: Is there a similar way to identify `subcases` in cocotb==1.6.0?

ktbarrett, 2021-11-03:
This is simply a regression, we did not change how arguments are passed to test functions by `TestFactory`. It would be nice to have a concrete example from you. Otherwise, I'll try to replicate it myself tonight.

marlonjames, 2021-11-03:
Using the example above, it's failing because

```py3
def run_test(dut, subcase=None):`
    print(subcase) # will work with cocotb==1.5.2, will fail with cocotb==1.6.0
```

is not an `async` function. If you change it to `async def run_test(...):` it works.

Perhaps we should catch that a plain function is passed in and raise a better error? This is the error in the example because the function returns `None`:
```
     0.01ns INFO     run_test failed
                     Traceback (most recent call last):
                       File "/home/marlon/git/cocotb/cocotb/regression.py", line 624, in _my_test
                         await function(dut, *args, **kwargs)
                     TypeError: object NoneType can't be used in 'await' expression
```

The provided example doesn't work with 1.5.2 for me, resulting in the error:
```
Failed to import module test_testfactory: TestFactory requires a cocotb coroutine
```

c-93, 2021-11-03:
Thanks again for your quick reply. Unfortunately, the async does not solve my issue.

Here is a minimal example, which works with cocotb==1.5.2, but does not work with cocotb==1.6.0.
```
import cocotb
from cocotb.result import TestFailure, TestSuccess
from cocotb.regression import TestFactory

@cocotb.test()
async def run_test(dut, subcase=None):	
	if not subcase:
		print("error: subcase is none for cocotb==1.6.0!")
		raise TestFailure()
	else:
		print ("success: subcase is ", subcase)
		raise TestSuccess()

subcases = ["Eve", "Alice", "Bob"]
factory = TestFactory(run_test)
factory.add_option("subcase", subcases)
factory.generate_tests()

```

marlonjames, 2021-11-03:
Interesting, it's the `cocotb.test()` decorator. If you remove that it passes.

c-93, 2021-11-03:
Correct. Thanks for your quick and productive support. My testcase is now working even with cocotb==1.6.0. :)

In my real-world code, I used the decorator to configure `timeout_time` and `timeout_unit` as follows:
`@cocotb.test(timeout_time=200000, timeout_unit="us")`

This resulted in the same behavior as the empty `@cocotb.test()` decorator.

c-93, 2021-11-04:
Just for documentation purposes:
You solved my question. Feel free to close this issue.
However, we found a bug in cocotb==1.6.0 when using the python decorator and the subcase string.

suzizecat, 2021-11-16:
Coming across this by chance. Isn't this issue due to the fact that using the `@cocotb.test()` registers the function as a test on its own, thus running it outside the test factory with, obviously, `None` as a parameter (perhaps before starting the tests generated by the test factory) ? 

On a very side note, be careful when using `not stuff` to test if `stuff` is `None`. In your example, if you have an empty subcase, it will be treated as a "None". You might prefer using `stuff is not None` instead.

ktbarrett, 2021-11-16:
@suzizecat is correct, you probably shouldn't be decorating test functions you plan on using in `TestFactory`s with `@cocotb.test()`. If you need to test a `None` case, add it to a `TestFactory.add_option` call.

```python
async def test_stuff(dut, option):
    ...

tf = TestFactory(test_stuff)
tf.add_option("option", [None, 1, 2, 3])
tf.generate_tests()
```

suzizecat, 2021-11-16:
@ktbarrett anticipating the question (and because I'm interested in this), is it possible in this configuration to pass `cocotb.test()` parameters to the test factory (here in example `timeout_time=200000, timeout_unit="us"` ) ?

ktbarrett, 2021-11-17:
The `TestFactory` does not support that. I don't think using `TestFactory` on a `@cocotb.test` decorated function does either based on what @c-93 said:

> In my real-world code, I used the decorator to configure timeout_time and timeout_unit as follows:
@cocotb.test(timeout_time=200000, timeout_unit="us")
>
> This resulted in the same behavior as the empty @cocotb.test() decorator.

It's a missing feature. `TestFactory` should be totally rethought IMO.

closed 2021-12-25 by github-actions[bot]