cocotb/cocotb #2766: Questa logging is out of order

raysalemi, 2021-11-06:
I've found a logging bug with regards to Questa.

Given the following test, I expect to see the same output for Icarus and Questa. But the Questa output is out of order:

```
class LogComp(uvm_component):
    async def run_phase(self):
        self.raise_objection()
        self.logger.debug("This is debug")
        self.logger.info("This is info")
        self.logger.warning("This is warning")
        self.logger.error("This is error")
        self.logger.critical("This is critical")
        self.drop_objection()

class LogTest(uvm_test):
    def build_phase(self):
        self.comp = LogComp("comp", self)

@cocotb.test()
async def log_test(_):
    """Write log messages to file with no streaming"""
    await uvm_root().run_test("LogTest")
```
Instead, I get the following:

# Icarus
These messages are in order:

```
     0.00ns INFO     Found test testbench.log_test
     0.00ns INFO     running log_test (1/1)
     0.00ns INFO     testbench.py(9)[uvm_test_top.comp]: This is info
     0.00ns WARNING  testbench.py(10)[uvm_test_top.comp]: This is warning
     0.00ns ERROR    testbench.py(11)[uvm_test_top.comp]: This is error
     0.00ns CRITICAL testbench.py(12)[uvm_test_top.comp]: This is critical
  1000.00ns INFO     log_test passed
```

# Questa
These messages are out of order
```
     0.00ns INFO     testbench.py(9)[uvm_test_top.comp]: This is info
     0.00ns WARNING  testbench.py(10)[uvm_test_top.comp]: This is warning
     0.00ns ERROR    testbench.py(11)[uvm_test_top.comp]: This is error
     0.00ns CRITICAL testbench.py(12)[uvm_test_top.comp]: This is critical
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:240  in _embed_sim_init                 Python interpreter initialized and cocotb loaded!
#      0.00ns INFO     Running on ModelSim for Questa-64 version 2021.3 2021.07
#      0.00ns INFO     Running tests with cocotb v1.5.2 from /home/rsalemi/.local/lib/python3.7/site-packages/cocotb
#      0.00ns INFO     Seeding Python random module with 1636220011
#      0.00ns INFO     Found test testbench.log_test
#      0.00ns INFO     Running test 1/1: log_test
#      0.00ns INFO     Starting test: "log_test"
#                      Description: Write log messages to file with no streaming
#   1000.00ns INFO     Test Passed: log_test
```

# Maybe a pyuvm bug?

The lack of `#` on the **pyuvm** messages suggests that maybe I'm doing something wrong in getting a logger for each of my components. So here is what I'm doing:

```
    def __init__(self, name):
        super().__init__(name) 
        uvm_root_logger = logging.getLogger('uvm')
        # Every object gets its own logger
        logger_name = self.get_full_name() + str(id(self))
        self.logger = uvm_root_logger.getChild(logger_name)
        self.logger.setLevel(level=logging.INFO)  # Default is to print INFO
        # We are not sending log messages up the hierarchy
        self.logger.propagate = False
        self._streaming_handler = logging.StreamHandler()
        self._streaming_handler.addFilter(SimTimeContextFilter())
        self._streaming_handler.setLevel(logging.NOTSET)
        # Make log messages look like UVM messages
        self._uvm_formatter = PyuvmFormatter(self.get_full_name())
        self.add_logging_handler(self._streaming_handler)It's possible that this is a 
```
The `PyuvmFormatter` looks like this:

```
class PyuvmFormatter(SimColourLogFormatter):
    def __init__(self, full_name):
        self.full_name = full_name
        super().__init__()

    def format(self, record):
        new_msg = f"{record.filename}({record.lineno})"
        new_msg += f"[{self.full_name}]: " + record.msg
        record.msg = new_msg
        return super().format(record)
```

# Summary
Could you tell me if this is a Questa issue (since it works in Icarus), a pyuvm issue (since I pyuvm has subloggers,) or a cocotb issue?

Thanks!

[logging_bug.tar.gz](https://github.com/cocotb/cocotb/files/7490966/logging_bug.tar.gz)
[icarus_output.txt](https://github.com/cocotb/cocotb/files/7490956/icarus_output.txt)
[questa_output.txt](https://github.com/cocotb/cocotb/files/7490957/questa_output.txt)

marlonjames, 2021-11-07:
cocotb's logger uses `logging.StreamHandler(sys.stdout)`, which appears to be captured by Questa and displayed with `#` prefixed.
`logging.StreamHandler()` uses `sys.stderr` by default:
https://docs.python.org/3/library/logging.handlers.html#logging.StreamHandler

raysalemi, 2021-11-07:
Thanks.  That solved it.  Fixed in **pyuvm** 2.1a

closed 2021-11-07 by raysalemi