cocotb/cocotb #2767: Parsing cocotb output

raysalemi, 2021-11-07:
Is there a recommended way to save **cocotb** output and parse it to see if any tests say `FAIL`?

ktbarrett, 2021-11-07:
`results.xml` outputs test results in jUnit format which includes failing tests. Do not parse the output, that may change at any time. See [bin/combine_results.py](https://github.com/cocotb/cocotb/blob/master/bin/combine_results.py) for an example of how to get failing tests from the results.xml file.

raysalemi, 2021-11-07:
Thank you, sir.

raysalemi, 2021-11-07:
I don't see the pass fail result in in this file:

```
<testsuites name="results">
  <testsuite name="all" package="all">
    <property name="random_seed" value="1636295397" />
    <testcase name="random_test" classname="testbench" file=".../testbench.py" lineno="134" time="0.003201007843017578" sim_time_ns="151000" ratio_time="47172642.9316252" />
    <testcase name="max_test" classname="testbench" file="...testbench.py" lineno="141" time="0.0017879009246826172" sim_time_ns="150000" ratio_time="83897266.30217363">
      <failure />
    </testcase>
  </testsuite>
</testsuites>
```

closed 2021-11-07 by raysalemi

Botnic, 2021-11-07:
You seem to have two test:
* random_test: Success
* max_test: Fail

raysalemi, 2021-11-07:
Yes, I found the <failure /> tag.