cocotb/cocotb #2773: cocotb-config failes due to errors in libgpi

aviadb, 2021-11-11:
Hi,
I installed cocotb version `1.6.0` using pip under my conda environment (python=3.9) on Fedora Linux x64.

When I execute `cocotb-config` I get the following error:

`
Traceback (most recent call last):
  File "/home/aviadb/miniconda3/envs/cocotb/bin/cocotb-config", line 5, in <module>
    from cocotb.config import main
  File "/home/aviadb/miniconda3/envs/cocotb/lib/python3.9/site-packages/cocotb/__init__.py", line 43, in <module>
    import cocotb.handle
  File "/home/aviadb/miniconda3/envs/cocotb/lib/python3.9/site-packages/cocotb/handle.py", line 39, in <module>
    from cocotb import simulator
ImportError: /home/aviadb/miniconda3/envs/cocotb/lib/python3.9/site-packages/cocotb/libs/libgpi.so: undefined symbol: _ZSt28__throw_bad_array_new_lengthv
`
Any ideas?

ktbarrett, 2021-11-11:
>undefined symbol: _ZSt28__throw_bad_array_new_lengthv

This is a C++ standard library symbol. Are you building with a different C++ standard library than you are loading at runtime? Which simulator are you using?

aviadb, 2021-11-11:
1. Are you building with a different C++ standard library than you are loading at runtime? - Not sure, I just used `pip install` as written in the installation instructions. 
2. Which simulator are you using? - Icarus Verilog

How can I know which std library I use?

closed 2022-01-03 by github-actions[bot]

WSEmma, 2022-01-25:
Has your question been resolved?I have met the same question

aviadb, 2022-01-25:
@WSEmma 
The problem is with miniconda that installs outdated libraries under the the environment.
You should point to your local library files.
I.E.  under your conda environment, find the faulty .so files, remove them and create a link to your local linux installation 
.so libs.