cocotb/cocotb #2778: First attempt with adder example gives a error in a path

Albert-Hoevenaars, 2021-11-18:
In a first attempt to see how cocotb works I copied the adder example. Place the files adder.vhd, adder_model.py, test_adder.py and the makefile into one directory. Opened Windows Powershell in that directory and  give the command "make". I get some error messages that I don't understand.
---------------
PS C:\Project_github\fdc_sib\pld\CoCoTB_testbenches\adder> make sim=questa                                                                      makefile:18: C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\share\makefilesMakefile.sim: No such file or directory                                                                                                                                          make: *** No rule to make target 'C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\share\makefilesMakefile.sim'.  Stop.     
-------------
I noticed that makefilesMakefile.sim is not correct. But adding a "\" or "/" in the path gives another error message.

Someone a solution?

Albert-Hoevenaars, 2021-11-18:
I made some changes to the makefile  and I am getting now:

# do sim_build/runsim.do
# QuestaSim-64 vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap -c 
# ** Warning: vmap will not overwrite local modelsim.ini.
# QuestaSim-64 vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap work sim_build/work 
# Modifying modelsim.ini
# QuestaSim-64 vcom 2020.2 Compiler 2020.04 Apr 19 2020
# Start time: 08:51:50 on Nov 18,2021
# vcom -work work "+acc" C:/Project_github/fdc_sib/pld/CoCoTB_testbenches/adder/adder.vhd 
# -- Loading package STANDARD
# -- Loading package TEXTIO
# -- Loading package std_logic_1164
# -- Loading package NUMERIC_STD
# -- Compiling entity adder
# -- Compiling architecture rtl of adder
# End time: 08:51:50 on Nov 18,2021, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -onfinish exit -foreign "cocotb_init C:/Users/320123287/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll" sim_build/work.adder 
# Start time: 08:51:50 on Nov 18,2021
# ** Note: (vsim-3812) Design is being optimized...
# //  Questa Sim-64
# //  Version 2020.2 win64 Apr 19 2020
# //
# //  Copyright 1991-2020 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading ieee.numeric_std(body)
# Loading work.adder(rtl)#1
# Loading C:/Users/320123287/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll
#      -.--ns INFO     cocotb.gpi                         ..ib\fli\FliImpl.cpp:1105 in cocotb_init                     cocotb_init called
# 
#      -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# 
#      -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:99   in gpi_print_registered_impl       FLI registered
#      0.00ns INFO     Running on ModelSim for Questa-64 version 2020.2 2020.04
# 
#      0.00ns INFO     Running tests with cocotb v1.6.0 from C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb
# 
#      0.00ns INFO     Seeding Python random module with 1637221912
# 
#      0.00ns WARNING  Pytest not found, assertion rewriting will not occur
# 
#      0.00ns CRITICAL Failed to import module test_adder: invalid syntax (adder_model.py, line 5)
# 
#      0.00ns INFO     MODULE variable was "test_adder"
# 
#      0.00ns INFO     Traceback: 
# 
#      0.00ns INFO     Traceback (most recent call last):
# 
#                        File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\regression.py", line 173, in _discover_tests
# 
#                          module = _my_import(module_name)
# 
#                        File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\regression.py", line 67, in _my_import
# 
#                          mod = __import__(name)
# 
#                        File "C:\Project_github\fdc_sib\pld\CoCoTB_testbenches\adder\test_adder.py", line 6, in <module>
# 
#                          from adder_model import adder_model
# 
#                        File "C:\Project_github\fdc_sib\pld\CoCoTB_testbenches\adder\adder_model.py", line 5
# 
#                          def adder_model(a int, b int) - int
# 
#                                            ^
# 
#                      SyntaxError: invalid syntax
# 
#                      
# 
#      0.00ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
# 

Would expect that the examples work very quickly and easily.

Somebody having a solution for this problem?

imphil, 2021-11-18:
Can you let us know a bit more about how you installed cocotb and what environment (windows, how you installed Python, etc.)?

Albert-Hoevenaars, 2021-11-18:
I followed the installation as mentioned on https://docs.cocotb.org/en/stable/install.html. I am a hardware designer and my knowledge of software and all kind of installations is rather poor. I found the installation description difficult to follow. Python I installed a while ago when I was working with MyHDL. No problem found with Pyhton3. Don't know what I should do with "Python development packages (Python/C API headers and embedding library)". Not clear to me. For "GCC 4.8.1+, Clang 3.3+ or Microsoft Visual C++ 14.21+ and associated development packages" I installed Visual Studio with C++ package. For "GNU Make 3+" I installed miniconda3 as proposed. I also executed the command in the Conda console. Installed cococtb according to the given command. As simulator I am using Questasim version 2020.2 and I am running everything under Windows 10. By my lats tests I see that Questasim is trying to start, but get some errors.

Albert-Hoevenaars, 2021-11-18:
Environment variable are:
C:\Users\xxxxxxxxx\Miniconda3\Library\usr\bin
C:\pldtools\questasim64_2020.2\win64

ktbarrett, 2021-11-18:
@Albert-Hoevenaars You might want to look at [this guide on GitHub markdown](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks).

Your error is a syntax error, see the stacktrace you posted
```
#      0.00ns INFO     Traceback (most recent call last):
# 
#                        File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\regression.py", line 173, in _discover_tests
# 
#                          module = _my_import(module_name)
# 
#                        File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\regression.py", line 67, in _my_import
# 
#                          mod = __import__(name)
# 
#                        File "C:\Project_github\fdc_sib\pld\CoCoTB_testbenches\adder\test_adder.py", line 6, in <module>
# 
#                          from adder_model import adder_model
# 
#                        File "C:\Project_github\fdc_sib\pld\CoCoTB_testbenches\adder\adder_model.py", line 5
# 
#                          def adder_model(a int, b int) - int
# 
#                                            ^
# 
#                      SyntaxError: invalid syntax
```

It should look like the below. Did you copy-paste incorrectly?
```python
def adder_model(a: int, b: int) -> int:
    ...
```

ktbarrett, 2021-11-18:
We call the `find` command in the makefiles. I'm guessing the call format we are using isn't compatible with Powershell's version of `find`, but it seems to work regardless. Ideally, we would update the usages of `find` to work with both PS and Bash.

Albert-Hoevenaars, 2021-11-23:
> Caution: This e-mail originated from outside of Philips, be careful for phishing.

Kaleb,

thanks for the message. Will ignore these "Parameter format not correct" messages.

Regards

Albert Hoevenaars

The information contained in this message may be confidential and legally protected under applicable law. The message is intended solely for the addressee(s). If you are not the intended recipient, you are hereby notified that any use, forwarding, dissemination, or reproduction of this message is strictly prohibited and may be unlawful. If you are not the intended recipient, please contact the sender by return e-mail and destroy all copies of the original message.

ktbarrett, 2021-11-26:
Seeing as `find` works and we are moving away from the makefiles, I am not considering the format warnings actionable, and with your questions answered i am closing the issue. If you feel that all your questions are not answered, feel free to reopen the issue.

closed 2021-11-26 by ktbarrett