cocotb/cocotb #2779: Enhancing the documentation on "accessing the design"

walt3k, 2021-11-20:
It would be nice if  
https://docs.cocotb.org/en/latest/writing_testbenches.html#accessing-the-design 
could elaborate on all the different syntax a cocotb testcase can use to access different entities in the model.  For example, I am opening this issue because I was looking for the exhaustive list of the different ways multidimensional arrays can be accessed and set.  I am sure there are other examples/scenarios which could also be added.  On gitter, @ktbarrett suggested an issue would be appropriate for such an enhancement request.

ktbarrett, 2021-11-20:
To best guide the improvement, what exactly do you think is missing or needs improvement?

walt3k, 2021-11-20:
I think showing an example how multi-dimensional arrays can be accessed would be helpful, for e.g

1. setting/initializing an entire multi-dimensional array
2. setting one row in the middle of a multi-dimensional array
3. setting one element in the middle of a row which is in the middle of a multi-dimensional array
4. other syntax/capabilities I am not thinking of?
5. show slicing example which is not supported

I will work on additional ideas.  I am a cocotb novice so I have a cart/horse problem since I don't know what don't know.  :-)  thx!