cocotb/cocotb #2780: Questa 64-bit invalid ELF header

Anasslm, 2021-11-22:
Hi,
I am trying to run cocotb and i got this issue. I have looked into this issue https://github.com/cocotb/cocotb/issues/2386 i think it's because of the .so file that no longer existing on the latest python version, i tried to use instead .a but i got this issue.

System:
* the cocotb version : 1.6.0
* Linux (64 bit),
* Mentor Questasim 2021.3  (64 bit),
* Python 3.9.9 (pip installer),

Output example

```
make -f Makefile results.xml
make[1]: Entering directory `cocotb_first_test'
set -o pipefail; MODULE=test_dff TESTCASE= TOPLEVEL="work.dff" \
GPI_EXTRA= TOPLEVEL_LANG=verilog \
 /software/mentor/questasim/questasim_2021.3/questasim/bin/vsim -c -64  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Reading pref.tcl

# 2021.3

# do sim_build/runsim.do
# QuestaSim-64 vmap 2021.3 Lib Mapping Utility 2021.07 Jul 13 2021
# vmap -c 
# ** Warning: vmap will not overwrite local modelsim.ini.
# QuestaSim-64 vmap 2021.3 Lib Mapping Utility 2021.07 Jul 13 2021
# vmap work sim_build/work 
# Modifying modelsim.ini
# QuestaSim-64 vlog 2021.3 Compiler 2021.07 Jul 13 2021
# Start time: 11:50:00 on Nov 22,2021
# vlog -work work "+define+COCOTB_SIM" -sv -timescale 1ns/1ps -mfcu "+acc" 
/cocotb_first_test/dff.sv 
# -- Compiling module dff
# 
# Top level modules:
#       dff
# End time: 11:50:00 on Nov 22,2021, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -onfinish exit -pli "/soft_lamp/python3/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_modelsim.so" sim_build/work.dff 
# Start time: 11:50:00 on Nov 22,2021
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vsim-3865) Due to PLI being present, global +acc is being enabled automatically. This will cause your simulation to run very slowly. Please use vsim -no_autoacc to disable this feature. This feature is now deprecated and will be removed from future releases.
# //  Questa Sim-64
# //  Version 2021.3 linux_x86_64 Jul 13 2021
# //
# //  Copyright 1991-2021 Mentor Graphics Corporation
# //  All Rights Reserved.

# //
# Loading sv_std.std
# Loading work.dff(fast)
# Loading /soft_lamp/python3/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_modelsim.so
#      -.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:68   in utils_dyn_open                  Unable to open lib soft_lamp/python3/lib/libpython3.9.a: /soft_lamp/python3/lib/libpython3.9.a: invalid ELF header
#      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
# End time: 11:50:01 on Nov 22,2021, Elapsed time: 0:00:01
# Errors: 0, Warnings: 1
ERROR: results.xml was not written by the simulation!
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory `cocotb_first_test'
make: *** [sim] Error 2
```
Any suggestions ?

Thanks

ktbarrett, 2021-11-22:
You cannot dynamically load an archive, it must be a shared object. I don't know which Python environment you are using, but look for an add-on that has libpython3.9.so, for example if you got this from your OS's package manager, there is likely a separate package called python3-dev or libpython3.

Also, check the output of `cocotb-config --libpython`, if it's the path to `libpython3.9.a`, please file a bug report against [`find_libpython`](https://github.com/ktbarrett/find_libpython) and give *a lot* of detail on your Python environment.

Anasslm, 2021-11-22:
Thanks for the reply, it's now working fine 
The output of cocotb-config --libpython is  libpython3.9.so

closed 2021-11-22 by Anasslm