cocotb/cocotb #2781: TestFactory: auto-generated test names are not applied

chdet, 2021-11-22:
System : 
Cocotb 1.6.0 
Python 3.9
Linux 64bits

The removal of these lines in the RegressionManager method  '_create_test' in commit fc32b3bb27bf78dd885c833e581f51bfeb50900b

>     _my_test.__name__ = name
>     _my_test.__qualname__ = name
>     _my_test.__doc__ = documentation
>     _my_test.__module__ = mod.__name__

Results in the generated tests all sharing the name of the base test function, e.g.

> Found test module.parametrized_test
> Found test module.parametrized_test
> Found test module.parametrized_test
> Found test module.parametrized_test
> etc ...

Instead of being append an index name as expected

> Found test module.parametrized_test_001
> Found test module.parametrized_test_002
> Found test module.parametrized_test_003
> Found test module.parametrized_test_004
> etc ...

ktbarrett, 2021-11-22:
Should be fixed. This will go out in cocotb v1.6.1 which may take a couple days to release.

imphil, 2021-12-07:
Fix for master (also includes test): #2788
Fix for 1.6.1: https://github.com/cocotb/cocotb/pull/2782. Test cherry-picked later in #2798

PR #2782 (2021-11-22): Fix regression in TestFactory wrt unique test names
Closes #2781. BTW the last line breaks https://github.com/cocotb/cocotb/pull/2721#discussion_r717707761. We just can't win with this stuff can we?
closed 2021-12-07 by imphil