cocotb/cocotb #2785: Using different compile libraries in Questa

Albert-Hoevenaars, 2021-11-23:
I am using HDL Designer to create big VHDL designs. Units from one library can used (from example) a package file from another library. Until now I compiled my design using a batch file where I compiled every design unit to its own library. Used "vmap" to create these libraries and "vcom" to compile every file I needed for the simulation to the vmapped library where it belongs to. Then a "vsim" command started the simulation. This works great.
With cocotb I want to compile a file from a library that wants to use another package file in the library, which gives an error.
In the file "aaa_pkg" (file is in lib_bbb) the following lines can be found:
   library lib_bbb;
   use lib_bbb.project_pkg.all;

The error message is get is that lib_bbb is not found. Is that because everything is complied to "work". If so, is there a work around (please not by changing all library names to "work").

Regards
 
Albert

ktbarrett, 2021-11-23:
Have a look [here](https://docs.cocotb.org/en/stable/building.html). The process you described is similar to what is done in the Questa makefiles, you just have to give it the right info.

Albert-Hoevenaars, 2021-11-23:
> Caution: This e-mail originated from outside of Philips, be careful for phishing.

Kaleb,

do you mean that I have to change the questa makefile. I had a look into it already and changing it is not an easy task (I am a hardware designer,
not a software designer). Can you give a quick example? Or do you means something else?

Thanks in advance.

Albert

The information contained in this message may be confidential and legally protected under applicable law. The message is intended solely for the addressee(s). If you are not the intended recipient, you are hereby notified that any use, forwarding, dissemination, or reproduction of this message is strictly prohibited and may be unlawful. If you are not the intended recipient, please contact the sender by return e-mail and destroy all copies of the original message.

Albert-Hoevenaars, 2021-11-23:
I tried several settings in my makefile but in the Powershell window I only see "vmap -c" and "vmap work simbuild/word". So no other directories are created.
Used in the make file "VHDL_LIB_ORDER"  but that does not create other libs. Also VHDL_SOURCES_<lib> does not help. 
Any help?? Maybe some example that will also be available for other people.

Albert-Hoevenaars, 2021-11-23:
By using the following makefile I managed to get 3 libraries compiled without any error message
------
#file: makefile
TOPLEVEL_LANG = vhdl
SIM = questa
WAVES = 1
COCOTB_LOG_LEVEL = TRACE

PWD = $(shell pwd)

MODULE = axi4wps_tester
TOPLEVEL = axi4wps_iif_wps
#
WPS_FILES = ../../fpga1/Axi4WpsLib/Axi4WpsLib/hdl
IPSHELL_FILES = ../../fpga1/ipShellLib/ipShellLib/hdl
FLASHLITE_FILES = ../../fpga1/FlashLiteLib/FlashLiteLib/hdl
#

VHDL_SOURCES_Axi4WpsLib = $(WPS_FILES)/axi4wps_regs_pkg.vhd $(WPS_FILES)/axi4wps_iif.vhd
VHDL_SOURCES_FlashLiteLib = $(FLASHLITE_FILES)/project_pkg.vhd $(FLASHLITE_FILES)/image_pkg.vhd
VHDL_SOURCES_ipShellLib = $(IPSHELL_FILES)/ipshell_i2f_image_pkg.vhd
#
VHDL_LIB_ORDER = FlashLiteLib ipShellLib Axi4WpsLib
#
#EXTRA_ARGS = 
#SIM_ARGS = -cover sbcef -coverNoSub -2008

include C:/Users/320123287/AppData/Local/Programs/Python/Python39/Lib/site-packages/cocotb/share/makefiles/Makefile.sim
#include C:/cocotb/cocotb/cocotb/share/makefiles/Makefile.sim
#include $(shell cocotb-config --makefiles)/Makefile.sim
---
When the vsim command is send to start Questasim it wants to start with sim_build/work.axi4wps_iif_wps, but this gives an error because it can not find the file.
------
# vmap work sim_build/work
# Modifying modelsim.ini
# vsim -onfinish exit -foreign "cocotb_init C:/Users/320123287/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll" sim_build/work.axi4wps_iif_wps
# Start time: 14:18:21 on Nov 23,2021
# ** Note: (vsim-3812) Design is being optimized...
# ** Error: (vopt-13130) Failed to find design unit axi4wps_iif_wps.
#         Searched libraries:
#             work
#             sim_build/work
# Optimization failed
----------------------------
What's going wrong?

ktbarrett, 2021-11-23:
> ```
> # vsim -onfinish exit -foreign "cocotb_init C:/Users/320123287/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll" sim_build/work.axi4wps_iif_wps
> ```

It's trying to load `work.axi4wps_iif_wps`, when presumably that entity is in the `Axi4WpsLib`? It's prepending the `RTL_LIBRARY` variable before the `TOPLEVEL` for some reason ([see here](https://github.com/cocotb/cocotb/blob/3c809e3e7f54512f6f7c865676febf0fe5f493e2/cocotb/share/makefiles/simulators/Makefile.questa#L47)). Trying changing `RTL_LIBRARY` to `Axi4WpsLib`. Remember to run `make clean` on your project if you make changes to the Makefile.

Albert-Hoevenaars, 2021-11-24:
After adding RTL_LIBRARY (could not find anything about this item in the documentation) the vcom command points to the correct library.
--
# vsim -onfinish exit -foreign "cocotb_init C:/Users/320123287/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll" sim_build/Axi4WpsLib.axi4wps_iif
# Start time: 08:04:56 on Nov 24,2021
# ** Note: (vsim-3812) Design is being optimized...
# ** Error: (vopt-13130) Failed to find design unit axi4wps_iif.
--
But again an error because it cannot find the design unit. All files are compiled without any error.

Albert-Hoevenaars, 2021-11-24:
When I start Questasim, change the work directory and execute the commands from the runsim.do, there are no compile errors. but again it cannot find axi4wps_iif. When I look in the libraries shown in Questasim the axi4wpslib is empty. the other two are correct and are showing files.
I had a closer look into the runsim.do and found the following:

if [file exists sim_build/Axi4WpsLib] {vdel -lib sim_build/Axi4WpsLib -all}
vlib sim_build/Axi4WpsLib
vmap Axi4WpsLib sim_build/Axi4WpsLib
vcom -work Axi4WpsLib +acc C:/Project_github/fdc_sib/pld/fpga1/Axi4WpsLib/Axi4WpsLib/hdl/axi4wps_regs_pkg.vhd C:/Project_github/fdc_sib/pld/fpga1/Axi4WpsLib/Axi4WpsLib/hdl/axi4wps_iif.vhd

if [file exists sim_build/Axi4WpsLib] {vdel -lib sim_build/Axi4WpsLib -all}
vlib sim_build/Axi4WpsLib
vmap Axi4WpsLib sim_build/Axi4WpsLib
vsim  -onfinish exit -foreign "cocotb_init C:/Users/320123287/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll"   sim_build/Axi4WpsLib.axi4wps_iif
log -recursive /*
onbreak resume
run -all
quit

It cleans the library axi4wpslib, defines a new one and maps it. Then files are compiled. But then it deletes the just created files in the aix4wpslib library, defines a new lib and maps it. But it does not compile again, instead it starts Questasim with a just cleaned aix4wpslib. I removed the 3 lines after the vcom for library aix4wpslib. Started Questasim and executed the do file. now the simulation is started, design is opened, although I see one error messages "cocotb.gpi                         ..s\cocotb_utils.cpp:56   in utils_dyn_open                  Unable to open lib python39.dll: The specified module could not be found." 

Why is the library created compiled and cleaned? Any remarks on the last error message?

Best Regards,

Albert

ktbarrett, 2021-11-24:
> Why is the library created compiled and cleaned?

After looking into the Makefile further, I see that my recommendation to change RTL_LIBRARY to Axi4WpsLib is not correct. RTL_LIBRARY is intended to be the library name for VHDL_SOURCES.

The problem is that the current makefile makes the incorrect assumption that TOPLEVEL is in RTL_LIBRARY. Instead we should add a TOPLEVEL_LIBRARY for you to specify the library the TOPLEVEL is in. We can default that variable to the value of RTL_LIBRARY for backwards compatibility.

Also, we should change the default value for RTL_LIBRARY to something that isn't `work`. That breaks the intended usage of `work` in VHDL. I suggest we change it to `default`.

> Any remarks on the last error message?

Which Python environment are you using? Does it contain a python39.dll? cocotb needs the Python dynamic library to function.

Albert-Hoevenaars, 2021-11-24:
Kaleb,

thanks for the answer, but I do not really understand what you are meaning.
How can I correct the makefile, so that it will work correctly? If I need to organize my libraries in another, please tell me.
Below my current makefile.
--
#file: makefile
TOPLEVEL_LANG = vhdl
SIM = questa
WAVES = 1
#
PWD = $(shell pwd)
#
RTL_LIBRARY = Axi4WpsLib
MODULE      = axi4wps_tester
TOPLEVEL    = axi4wps_iif
#
WPS_FILES       = ../../fpga1/Axi4WpsLib/Axi4WpsLib/hdl
IPSHELL_FILES   = ../../fpga1/ipShellLib/ipShellLib/hdl
FLASHLITE_FILES = ../../fpga1/FlashLiteLib/FlashLiteLib/hdl
#
VHDL_SOURCES_Axi4WpsLib   = $(WPS_FILES)/axi4wps_regs_pkg.vhd $(WPS_FILES)/axi4wps_iif.vhd
VHDL_SOURCES_FlashLiteLib = $(FLASHLITE_FILES)/project_pkg.vhd $(FLASHLITE_FILES)/image_pkg.vhd
VHDL_SOURCES_ipShellLib   = $(IPSHELL_FILES)/ipshell_i2f_image_pkg.vhd
#
VHDL_LIB_ORDER = FlashLiteLib ipShellLib Axi4WpsLib
#
include C:/Users/320123287/AppData/Local/Programs/Python/Python39/Lib/site-packages/cocotb/share/makefiles/Makefile.sim
--

The python39.dll file was found on 3 places
***@***.***

Many thanks for all the help you gave me.

Best Regards,

Albert Hoevenaars

ktbarrett, 2021-11-24:
Changes are going to have to be made to the Questa makefiles to support your use case. Right now you can copy Makefile.questa and make the following changes:

```diff
--- a/cocotb/share/makefiles/simulators/Makefile.questa
+++ b/cocotb/share/makefiles/simulators/Makefile.questa
@@ -44,7 +44,8 @@ endif

 RTL_LIBRARY ?= work

-TOPLEVEL := "$(RTL_LIBRARY).$(TOPLEVEL)"
+TOPLEVEL_LIBRARY ?= $(RTL_LIBRARY)
+TOPLEVEL := "$(TOPLEVEL_LIBRARY).$(TOPLEVEL)"

 ifndef VLOG_ARGS
     VLOG_ARGS = -timescale $(COCOTB_HDL_TIMEUNIT)/$(COCOTB_HDL_TIMEPRECISION) -mfcu
```

Then instead of setting `RTL_LIBRARY` in your Makefile, you should set `TOPLEVEL_LIBRARY`.

You can also forsake the Makefiles entirely, and integrate cocotb into your existing workflow. See [this documentation](https://docs.cocotb.org/en/stable/custom_flows.html).

Albert-Hoevenaars, 2021-11-25:
Kaleb, thanks for helping me again. I made the modifications in the makefile.questa and they work. Questasim starts and uses the correct toplevel. Simulation runs with a few warnings.
1) Message:  Did not detect Python virtual environment. Using system-wide Python interpreter.
     There is a path in the environment variables:    C:\Users\320123287\AppData\Local\Programs\Python\Python39\
     Is that not enough?
2)  Message: Pytest not found, assertion rewriting will not occur.

The second solution that you mentioned in your comment, needs some time. I first want to do some experiments with the combination of Python and VHDL. Especially creating models in Python.

ktbarrett, 2021-11-25:
Warning 1 can be ignored. That is specifically stating whether Python virtual environments (see [virtualenv](https://virtualenv.pypa.io/en/latest/) and [venv](https://docs.python.org/3/library/venv.html)) are being used, or if cocotb was installed globally. Warning 2 is stating that because the pytest package is not also installed, some additional features are not supported. I recommend installing pytest, but you can ignore that warning as well.

Albert-Hoevenaars, 2021-11-25:
Kaleb,

thank you again for the marvelous support. You helped me a lot.

Best Regards

Albert Hoevenaars

PR #2787 (2021-12-10): Support TOPLEVEL that's not in RTL_LIBRARY in Questa makefiles
Closes #2785.
closed 2021-12-10 by ktbarrett