cocotb/cocotb #2786: gpi_get_handle_by_name faild

captainliuy, 2021-11-25:
# Host Env

* cocotb version  = 1.6.0,
* os version = centos7.6 x86_64,
* simulator and version = Questa Sim-64 vsim 10.7c,
* Python version = 3.6.8

## Step 1

[axil_ram](https://github.com/alexforencich/verilog-axi/tree/master/tb/axil_ram) test passed

## Step 2

Comments the signal in [s_axil_awprot](https://github.com/alexforencich/verilog-axi/blob/master/rtl/axil_ram.v#L50) in axil_ram.v test passed
below is part of the debug info:

```
INFO : VPI registered
DEBUG: Signal name s_axil_awprot
DEBUG: Searching for s_axil_awprot
DEBUG: Checking if s_axil_awprot is native through implementation VPI
DEBUG: Unable to query vpi_get_handle_by_name axil_ram.s_axil_awprot
DEBUG: Failed to find a handle named s_axil_awprot via any registered implementation
```

## Step 3

a mixed lang desgin, a verilog top contain a VHDL instance. axilite signal `awprot` not in the top port list

```
INFO : VPI registered
INFO : FLI registered
DEBUG: Signal name s_axi_ctrl_awprot
DEBUG: Searching for s_axi_ctrl_awprot
DEBUG: Checking if s_axi_ctrl_awprot is native through implementation VPI
DEBUG: Unable to query vpi_get_handle_by_name u0.s_axi_ctrl_awprot
DEBUG: Checking if s_axi_ctrl_awprot is native through implementation FLI
DEBUG: Looking for child s_axi_ctrl_awprot from u0
ERROR: FATAL: We are calling up again
```

captainliuy, 2021-11-25:
so if both VPI and FLI registered,  a optional verilog signal not found through VPI, then use the FLI, fatal error occurs

ktbarrett, 2021-11-26:
Are you seeing a stacktrace or anything else? Some changes were recently made to improve Questa's mixed language support #2776. Could you try using cocotb's master to see if you still have these issues?

You can install cocotb's current development version using 
```
pip install git+https://github.com/cocotb/cocotb.git@master
```

captainliuy, 2021-11-26:
Great! bug fixed in 1.7.0.dev0

closed 2022-03-11 by captainliuy