cocotb/cocotb #2795: Extending existing build flows

Albert-Hoevenaars, 2021-12-01:
I am running the latest version of cocotb and I am trying to following the description given here https://docs.cocotb.org/en/stable/custom_flows.html?highlight=extending%20existing.
I am using Questasim with VHDL files and starting Questasim with a batch file that creates directories, compiles everything I need in my design and then starts Questasim with the vsim command. So far so good.

I do not understand what I have to do with the extra command that I should add to the vsim command. Should I copy the give line -foreign "cocotb_init $(cocotb-config --lib-name-path fli questa)" literally to the vsim command line? I get an error message on this (Unable to expand "$cocotb")

Is it OK if I use in the batchfile a set command like: "set LIBPYTHON_LOC=C:\Users\xxxxxx\AppData\Local\Programs\Python\Python39\python39.dll" and sonting the same for set MODULE=....?

I also have some verilog IP blocks from Xilinx in my design. Unclear to me what I should do with the environment varibale "$(cocotb-config --lib-name-path vpi questa):cocotbvpi_entry_point".

Could someone help me with this?

Albert-Hoevenaars, 2021-12-01:
After carefully looking to the given commands (again), I think I already found out. Open a command window (in Windows) en give the command "cocotb-config --lib-name-path fli questa". I got the answer like this: "C:/Users/xxxx/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll"

The give the command (for verilog files): "cocotb-config --lib-name-path vpi questa" and the answer is "C:/Users/xxxx/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll" Use the answers in the vsim command and in a set command like this (hopefully this is all correct):

set LIBPYTHON_LOC=C:\Users\xxxx\AppData\Local\Programs\Python\Python39\python39.dll
set GPI_EXTRA=C:/Users/xxxx/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll

and add to the vsim command (including the ")
-foreign "cocotb_int C:/Users/xxxx/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll".

ktbarrett, 2021-12-01:
Setting up a custom flow is for advanced users who want to either integrate with an existing tool automation system, or find the Makefiles that cocotb provides inadequate. You are expected to figure the details out yourself.

>Should I copy the give line -foreign "cocotb_init $(cocotb-config --lib-name-path fli questa)" literally to the vsim command line?

This syntax only works in bash. See the note at the top of the documentation page. You need to run the command `cocotb-config --lib-name-path fli questa`, and use the output in the vsim command. There is probably some equivalent to the listed bash syntax in your shell, but I don't know what it is.

> Is it OK if I use in the batchfile a set command like: "set LIBPYTHON_LOC=C:\Users\xxxxxx\AppData\Local\Programs\Python\Python39\python39.dll" and sonting the same for set MODULE=....?

Sure. As long as they end up in the environment when `vsim` is run, it will work.

> I also have some verilog IP blocks from Xilinx in my design. Unclear to me what I should do with the environment varibale "$(cocotb-config --lib-name-path vpi questa):cocotbvpi_entry_point".

From the documentation:
> Set the GPI_EXTRA environment variable to `$(cocotb-config --lib-name-path vpi questa):cocotbvpi_entry_point` if there are also (System)Verilog modules in the design.

You can probably set this in the same batchfile as LIBPYTHON_LOC.

ktbarrett, 2021-12-01:
I would be cautious of hardcoding paths, the commands we provided are there to be used to dynamically evaluate those paths so the project is relocatable. If Windows command language doesn't support that use, I would not use it. Windows now ships with Powershell which is a far more featureful shell.

Albert-Hoevenaars, 2021-12-02:
Still see a message passing by in Questasim:
# ** Warning: (vsim-FLI-3159) Failed to find foreign initialization function 'cocotb_int' in FLI object file "C:/Users/320123287/AppData/Local/Programs/Python/Python39/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll".
Not sure if this is a problem.

Albert-Hoevenaars, 2021-12-02:
[error.txt](https://github.com/cocotb/cocotb/files/7640746/error.txt)

ktbarrett, 2021-12-02:
Unfortunately, you cannot import a module by giving it the full path. [See the Python documentation on how modules are discovered and imported.](https://docs.python.org/3/reference/import.html) In you case you will need to set PYTHONPATH, then set MODULE to a python import name. For example:

```batch
set PYTHONPATH="C:\\Project_github\\fdc_sib\\pld\\Questasim_batch\\"
set MODULE=axi4wps_tester
```

Typically you will set the PYTHONPATH to your project root and then can set MODULE to `path.to.test`.

Albert-Hoevenaars, 2021-12-02:
> Caution: This e-mail originated from outside of Philips, be careful for phishing.

Kaleb,

Yes this works. I see now one clock in my waveform.
But if I generate 2 clocks Questasim hangs
--
  cocotb.start_soon(Clock(dut.ip_clk, 10, units='ns').start())
  cocotb.start_soon(Clock(dut.axi4clk, 8, units='ns').start())
--
Any suggestion?

Thanks again

The information contained in this message may be confidential and legally protected under applicable law. The message is intended solely for the addressee(s). If you are not the intended recipient, you are hereby notified that any use, forwarding, dissemination, or reproduction of this message is strictly prohibited and may be unlawful. If you are not the intended recipient, please contact the sender by return e-mail and destroy all copies of the original message.

ktbarrett, 2021-12-02:
That should work. Your problem is likely elsewhere. Could you post the full code?

Albert-Hoevenaars, 2021-12-07:
[axi4wps_tester.txt](https://github.com/cocotb/cocotb/files/7666176/axi4wps_tester.txt)

Sorry for the delay. I am only working 3 days a week. Chnage ".txt" to ".py"

Albert-Hoevenaars, 2021-12-07:
If I use the following construction:
----
  c1 = Clock(dut.ip_clk, 10, 'ns')
  await cocotb.start(c1.start())
----
Questasim stays loading forever

If I use:
-----
  cocotb.start_soon(Clock(dut.ip_clk, 10, units='ns').start())
  cocotb.start_soon(Clock(dut.axi4clk, 8, units='ns').start())
---
Questasim stays also loading forever. Using one clock statement of the above 2, then Questasim starts normally

Albert-Hoevenaars, 2021-12-07:
This definition gives me 2 clock signals in Questasim:
---------
  c1 = Clock(dut.ip_clk, 10, 'ns')
  ip_clk = cocotb.start_soon(c1.start())
  c2 = Clock(dut.axi4clk, 8, 'ns')
  axi4clk = cocotb.start_soon(c2.start())
----------------

Regards

Albert

ktbarrett, 2021-12-07:
```python
        await RisingEdge(ip_clk)
```
Is a syntax error. `ip_clk` is not defined. There should be an error in either your transcript window or wherever stdout goes, presumably your terminal window.

Albert-Hoevenaars, 2021-12-07:
Kaleb,
although you helped me a lot, I am almost at a point to give up. I do not get any error about the await RisingEdge(ip_clk) statement. I added some more lines to my tester, but on a regular basis Questasim hangs (continues loading...) . When it runs it stops at 45 us. Possible due to the wrong waiting statement, but I am not sure. Below is my tester code:
----------------
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge,FallingEdge,Timer
from cocotb.result import TestFailure
from axi4wps_model import wps_model
#import logging
#import random

@cocotb.test()
async def wps_test(dut):

  c1 = Clock(dut.ip_clk, 6.4, 'ns')
  ip_clk = cocotb.start_soon(c1.start())
  c2 = Clock(dut.axi4clk, 6.4, 'ns')
  axi4clk = cocotb.start_soon(c2.start())
  
  #cocotb.start_soon(Clock(dut.ip_clk, 10, units='ns').start())
  #cocotb.start_soon(Clock(dut.axi4clk, 8, units='ns').start())
  #await cocotb.start(c1.start())
  #c2 = Clock(dut.axi4clk, 11, 'ns')
  #await cocotb.start(c2.start())
  #await cocotb.start(tb.reset_dut(dut.ip_rst, 20))

  """ 
    wait for some time to set registers 
  """
  await Timer(45, units='us')
  #
  """ wait for rising edge to start """
  #
  while (True):
    await RisingEdge(dut.ip_clk)
    """
      VHDL is now doing its thing
    """
    """
      get result values from dut (VHDL) after falling edge
    """
    await FallingEdge(dut.ip_clk)
    """
      get value from dut (VHDL)
    """
    py_wps_factor = dut.wps_factor.value
    #dut.wps_factor._log.info(py_wps_factor.binstr)
    
    py_wps_bypass = dut.wps_bypass.value
#    py_image_to_ip_format = dut.image_to_ip_format.value
#    py_image_to_ip_data = dut.image_to_ip_data.value 
#    py_image_to_ip_defect = dut.image_to_ip_defect.value
#    py_wait_to_ip = dut.wait_to_ip.value
    """ 
      send values to model
    """
#    wps_model(py_wps_factor, py_wps_bypass, py_image_to_ip_format, py_image_to_ip_data, py_image_to_ip_defect, py_wait_to_ip)
    """
      read values from model and move them to dut (VHDL)
    """  
#    dut.image_from_model_format.value = image_from_model_format
#    dut.image_from_model_data.value   = image_from_model_data
#    dut.image_from_model_defect.value = image_from_model_defect
#    dut.wait_from_model.value         = wait_from_model
    
    
    # wait some time
    await Timer(100, units='us')
#  

[... 53 lines cut ...]
    self.log.info(f"Found test {test.__module__}.{test.__qualname__}")
Message: 'Found test axi4wps_tester.wps_test'
Arguments: ()
--- Logging error ---
Traceback (most recent call last):
  File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\Lib\logging\__init__.py", line 1086, in emit
    stream.write(msg + self.terminator)
OSError: [Errno 9] Bad file descriptor
Call stack:
  File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\__init__.py", line 242, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\__init__.py", line 345, in _initialise_testbench_
    regression_manager.execute()
  File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\regression.py", line 462, in execute
    return self._start_test()
  File "C:\Users\320123287\AppData\Local\Programs\Python\Python39\lib\site-packages\cocotb\regression.py", line 471, in _start_test
    self.log.info(
Message: 'running wps_test (1/1)'
Arguments: ()
--------------
I appreciate your help very much. If you willing to help me further then I am very pleased.

Best Regards

Albert Hoevenaars

ktbarrett, 2021-12-07:
Oh I see your problem

```python
c1 = Clock(dut.ip_clk, 6.4, 'ns')
ip_clk = cocotb.start_soon(c1.start())
```
`ip_clk` is not a signal, it is a task, you cannot `RisingEdge(ip_clk)`. `RisingEdge` *only* works on signals. You should use `RisingEdge(dut.ip_clk)`.

cmarqu, 2021-12-07:
Try with `while (True):` changed to `while True:`.

Albert-Hoevenaars, 2021-12-08:
cmarqu, thanks for the replay. But I already tried to remove the brackets around the "True". Does not change anything.

Albert-Hoevenaars, 2021-12-08:
I noticed that when Questasim tries to start, which takes ages I see this message in the command window:
-----------
        1 file(s) copied.
Fatal Python error: init_sys_streams: can't initialize sys standard streams
Python runtime state: core initialized
OSError: [WinError 6] The handle is invalid

Current thread 0x000023cc (most recent call first):
<no Python frame>
----------------
I already checked the environment variable for user and system and there are two paths defined (in user and in system). One to the Python directory itself and another to the script directory.

Anybody having a solution for this error?

ktbarrett, 2021-12-08:
> Anybody having a solution for this error?

Questa on Windows has this issue. This is an upstream problem. We aren't aware of a workaround.

Albert-Hoevenaars, 2021-12-08:
Does this mean that I cannot use cocotb on windows? Spend already two weeks to get things working. :-(

ktbarrett, 2021-12-08:
You can use cocotb on Windows. Windows just makes it harder. That error message is an annoyance but IIRC isn't a cause of problems that prevent Questa from running (see #2015). If you want to use Questa on Windows you'll have to live with it until Siemens does something about it.

If you are willing to switch to Linux you can have things running in no time at all.

Albert-Hoevenaars, 2021-12-09:
To all who helped me the last couple of weeks: 
I created a small example to see if I get this running ( the other design has possible to many libraries and files, using VHDL and UVVM blocks). I managed to create a design with a counter (VHDL), then passed its values to a piece of Python that add 1 to the counter value. Then pushed the added value back into a decoder (VHDL). This design runs nicely in Questasim.  What I learned was getting values from VHDL and use them in Python and pushing values from Python back into VHDL.
What I learned from this example, I used that into my big design. But running this design with Questasim gives me a lot of errors. A lot of times the standard output does not seem to work, or I get a Questasim 211 error. The design stays continuously loading in Questasim due to a "can't initialize sys standard streams". Or the clock is not running, which gives a stop in my design. 
Only a few times the simulation was running. Even pushing fixed values from Python to VHDL does not work in the big design, while it was working in the small design.
All in all cocotb is for me a non reliable environment. I am using Windows as an OS and Linux is not an option.
For me is this the end of cocotb!

closed 2021-12-11 by ktbarrett