cocotb/cocotb #2797: Mixed-case TOPLEVEL causes FLI get_root_handle error

jonpovey, 2021-12-05:
If I supply mixed-case `TOPLEVEL=MyModule`, running the test fails with

```
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:509  in get_root_handle        FLI: Couldn't find root handle MyModule
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:515  in get_root_handle        FLI: Toplevel instances: MyModule != mymodule...
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:515  in get_root_handle        FLI: Toplevel instances: MyModule != standard...
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:515  in get_root_handle        FLI: Toplevel instances: MyModule != textio...
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:515  in get_root_handle        FLI: Toplevel instances: MyModule != std_logic_1164...
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:515  in get_root_handle        FLI: Toplevel instances: MyModule != std_logic_arith...
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:515  in get_root_handle        FLI: Toplevel instances: MyModule != std_logic_unsigned...
# 0.00ns ERROR  cocotb.gpi         FliImpl.cpp:515  in get_root_handle        FLI: Toplevel instances: MyModule != numeric_std...
# 0.00ns ERROR  cocotb.gpi       GpiCommon.cpp:282  in gpi_get_root_handle    No root handle found
# 0.00ns ERROR  cocotb.gpi       gpi_embed.cpp:324  in _embed_sim_init        cocotb initialization failed - exiting
# 0.00ns ERROR  cocotb.scheduler   __init__.py:302  in _sim_event             Failing test at simulator request before test run completion: Simulator shutdown prematurely
```
If I supply `TOPLEVEL=mymodule`, it passes. `TOPLEVEL=wrongtmodule` fails with a different error from the simulator:
```
# ** Error: (vsim-3170) Could not find 'wrongmodule'.
```

This is with a VHDL module that is written in mixed-case MyModule.

Info on my setup:

- cocotb  1.5.2
- Ubuntu 20.04 64-bit
- Modelsim DE 2019.1 32-bit (yes, 32-bit...)
- Python 3.7.1 under a miniconda 32-bit environment

ktbarrett, 2021-12-06:
Questa and the 32-bit ModelSim are different simulators. It would be interesting to see if this triggers in Questa as well. Hopefully this problem will become obsolete in the near future with Siemens putting out freely available rate-limited versions of Questa [like it is currently doing with Quartus](https://www.intel.com/content/www/us/en/software/programmable/quartus-prime/questa-edition.html).

jonpovey, 2021-12-06:
I also tested in `GHDL 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]` (in 64-bit conda environment and cocotb 1.6.0) with similar results. Works with lowercase, with mixed case I get:

```
     0.00ns INFO     Running tests with cocotb v1.6.0 from /home/jonp/miniconda3/envs/coco64/lib/python3.9/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1638755857
     0.00ns ERROR    VPI: Couldn't find root handle MyModule
     0.00ns ERROR    VPI: Toplevel instances: MyModule != mymodule...
     0.00ns ERROR    No root handle found
Traceback (most recent call last):
  File "/home/jonp/miniconda3/envs/coco64/lib/python3.9/site-packages/cocotb/__init__.py", line 242, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "/home/jonp/miniconda3/envs/coco64/lib/python3.9/site-packages/cocotb/__init__.py", line 320, in _initialise_testbench_
    raise RuntimeError(f"Can not find root handle ({root_name})")
RuntimeError: Can not find root handle (MyModule)
     0.00ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
```

marlonjames, 2021-12-07:
When using VHPI, we compare using the `CaseName` property, so the mixed-case should work there.

I don't know if there is a `CaseName` alternative to `mti_GetRegionName()`, but that would be the equivalent for FLI.

jonpovey, 2021-12-07:
AIUI VHDL is case-insensitive here, so it seems like this should be a case-insensitive comparison (force TOPLEVEL to lowercase)

richardc87, 2023-05-12:
I am also seeing this same issue with Questa, fails with same error with mixed case.  
Setup

- cocotb 1.7.2
- questa 2020.2 
- python 3.6.8
- Red Hat 7.9

ktbarrett, 2023-05-14:
This seems like something Questa should fix, no? Its inconsistent with the behavior of subhandles isn't it?

But I do understand why it may be this way. All objects in Questa (at least in the waveform viewer) are normalized to lowercase.