cocotb/cocotb #2800: Investigate threading issues in logger

ktbarrett, 2021-12-08:
From this comment
https://github.com/cocotb/cocotb/blob/cb876412fdaee249407619c1d4e34d3c04241754/cocotb/share/lib/gpi_log/gpi_logging.cpp#L114-L116

We aren't convinced this is correct. This may cause logs to clobber in the static buffer. This needs some investigation, and perhaps a mutex lock.

stezskal, 2021-12-09:
I have a behavior where when I set **COCOTB_REDUCED_LOG_FMT to 0**, the log output stops at 
```
46.87ns ERROR    cocotb.coroutine.Task 5.0xe8cc1d0c         scheduler.py:539  in _unschedule                     Exception raised by this forked coroutine
```

But when I set **COCOTB_REDUCED_LOG_FMT to 1**, I get a full traceback as one would desire:
```
   46.87ns INFO     tb_cocotb_listening_down_convert failed
                     Traceback (most recent call last):
                     ...
                     OverflowError: Int value (73728) out of range for assignment of 7-bit signal ('sb_addr')
```

Is this related to this threading issue?

ktbarrett, 2021-12-09:
No that is a different issue. This issue only *potentially* affects logs from C code. I would open a separate issue for that.

ktbarrett, 2025-02-17:
I am now convinced this is not an actual issue. Multiple threads are currently not a concern for the GPI.

closed 2025-02-17 by ktbarrett