cocotb/cocotb #2803: [tracking] Completion of VHPI support in Questa

imphil, 2021-12-09:
Questa is working on supporting VHPI in addition to FLI (which continues to work and is fully supported).

The current 2021.4 release shows the following test failures in the cocotb test suite:

```
Failure in testsuite: 'all' classname: 'test_vhdl_access' testcase: 'check_objects' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_array' testcase: 'test_gen_loop' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_array' testcase: 'test_discover_all' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_vhdl_zerovector' testcase: 'test_long_signal' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_vhdl_zerovector' testcase: 'test_read_zero_signal' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_vhdl_zerovector' testcase: 'test_write_zero_signal_with_0' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_vhdl_zerovector' testcase: 'test_write_zero_signal_with_1' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_force_release' testcase: 'test_force_release' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_iteration' testcase: 'recursive_discovery' with parameters 'all'
```

Plus #2749, which is expected to be fixed by Questa 2022.1 (to be confirmed), but will also be fixed by #2750 on the cocotb side.

Let's use this issue to track progress towards a full VHPI implementation (as far as cocotb needs it), and then remove the experimental label (https://github.com/cocotb/cocotb/blob/master/documentation/source/simulator_support.rst#mentorsiemens-eda-questa).

Steps to reproduce these failures:

```
git clone https://github.com/cocotb/cocotb.git
git checkout cb876412fdaee249407619c1d4e34d3c04241754 # at the time of writing, but also try master!
VHDL_GPI_INTERFACE=vhpi SIM=questa TOPLEVEL_LANG=vhdl tox -e py3
```

raysalemi, 2021-12-09:
I've shared these with Ritesh. 

 @imphil, could you open a Service Request to track this?

marlonjames, 2021-12-11:
Is the `vhpiCbRepStartOfPostponed` callback on the Questa roadmap?

I know they are planning on being more permissive in `vhpiCbRepEndOfTimeStep` callbacks, but I would rather be VHPI compliant if possible.

raysalemi, 2021-12-11:
> Is the `vhpiCbRepStartOfPostponed` callback on the Questa roadmap?
> 
> 
> 
> I know they are planning on being more permissive in `vhpiCbRepEndOfTimeStep` callbacks, but I would rather be VHPI compliant if possible.

Is this the restart issue?

raysalemi, 2021-12-11:
I'm not sure if @imphil has opened an SR on this issue.  If so, I could use the number.

marlonjames, 2021-12-12:
> Is this the restart issue?

No, this relates to #2749

raysalemi, 2021-12-12:
@Riteshmentor is driving these issues

raysalemi, 2022-04-01:
Thanks!  Are these new issues in the new version or are they simply things we haven't fixed yet?

ktbarrett, 2022-04-01:
@raysalemi I reported this a couple months ago. It is a bug, edge triggers are not working properly.

marlonjames, 2022-04-01:
`test_vhdl_access` is having an issue with `for generate` statements as well.

imphil, 2022-07-04:
Status update: The remaining issues should be fixed as part of an upcoming Questa release together with https://github.com/cocotb/cocotb/pull/3021 on the cocotb side.

imphil, 2022-07-05:
All code updates should be in cocotb now. Let's wait for the upcoming Questa release and close this issue if we can verify the functionality with the released version.

imphil, 2022-08-24:
Questa 2022.3 passes the cocotb test suite on master (as of today). https://github.com/cocotb/cocotb/pull/3055 documents this fact.

imphil, 2022-08-24:
cocotb CI is now also running against Questa 2022.3 and requires passing Questa VHPI tests from now on.

imphil, 2022-08-25:
The documentation is now updated as well, let's consider this task done!

closed 2022-08-25 by imphil