cocotb/cocotb #2804: Simulator crashes if forked coro raises an early exception

jonpovey, 2021-12-09:
This test causes Modelsim to crash with a SIGSEGV. @ktbarrett reports it is reproducible on master with Icarus too.
```
@cocotb.test()
async def bad_thread(dut):
    async def cr():
        while True:
            print(nonexistent)
            await Timer(1, 'ns')

    if 1:
        cocotb.fork(cr())
    else:
        cocotb.start_soon(cr())
    for n in range(100):
        await Timer(1, 'ns')
```

If `start_soon` is selected instead of `fork`, I get a graceful error messages instead:

```
#      0.00ns INFO     running bad_thread (1/1)
#      0.00ns ERROR    Exception raised by this forked coroutine
#      1.00ns INFO     bad_thread failed
#                      Traceback (most recent call last):
#                        File "/home/jonp/work/hot_coco/fifo_pyuvm/fifo_cocotb.py", line 103, in cr
#                          print(nonexistent)
#                      NameError: name 'nonexistent' is not defined
```

System here: Ubuntu 64-bit, 32-bit conda env, Modelsim DE 32-bit. VHDL.

ktbarrett, 2021-12-09:
The full exception:
```
Traceback (most recent call last):
  File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 587, in _resume_coro_upon
    trigger.prime(self._react)
  File "/home/kaleb/dev/cocotb/cocotb/triggers.py", line 655, in prime
    callback(self)
  File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 359, in _react
    self._event_loop(trigger)
  File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 453, in _event_loop
    self._schedule(coro, trigger=trigger)
  File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 891, in _schedule
    assert result is None
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kaleb/dev/cocotb/cocotb/__init__.py", line 242, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "/home/kaleb/dev/cocotb/cocotb/__init__.py", line 345, in _initialise_testbench_
    regression_manager.execute()
  File "/home/kaleb/dev/cocotb/cocotb/regression.py", line 472, in execute
    return self._start_test()
  File "/home/kaleb/dev/cocotb/cocotb/regression.py", line 493, in _start_test
    cocotb.scheduler._add_test(self._test_task)
  File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 784, in _add_test
    self._resume_coro_upon(
  File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 590, in _resume_coro_upon
    self._trigger2coros.pop(trigger)
KeyError: <NullTrigger for Start <Test 0> at 0x7f153ae4ddc0>
     0.00ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
```

When the `fork` call fails it `_unschedule`s the `cr` task. Because no `join` existed for that task, the test was ended as a failure. However, because `fork` is re-entrant, after the `fork` call exits, the test function *continues running*. Later it yields the `Timer` trigger. This causes the first internal assertion to occur. This is caught by the `_resume_coro_upon` that scheduled the test. Then `_resume_coro_upon` tries to remove the trigger that caused the failure from the `_trigger2coros` map, but it's empty since the `_event_loop` popped it off, causing the second exception. This causes the entry function `_initial_testbench` to exit with failure, and we know that the simulation is not cleaned up properly if that happens (#2018).

Complicated error for a 10 line test...

ktbarrett, 2021-12-09:
Duplicate of #1688, though it manifests in a slightly different way now that we do more internal checking in the scheduler.

closed 2021-12-09 by ktbarrett