cocotb/cocotb #2811: The 'external' decorator has regressed in functionality since v1.0

leftink, 2021-12-13:
The issue is present as of COCOTB v1.6.1.

The issue appears to be introduced in COCOTB v1.1.

In COCOTB v1.0, the external behaves the way it was intended, i.e. a thread is created and allowed to run will a Python Event trigger is provided to the scheduler to notify when the thread is done.  The scheduler was able to proceed and allow simulation to progress and the next time the event loop was run, if the thread had completed, the coroutine that was waiting on the external would then execute.

In the current implementation of the 'external' decorator, the scheduler blocks until all external threads have completed.  It doesn't even start all the external threads and then wait for all threads to complete.  The current implementation actually buys you nothing, i.e. I would essentially get the exact same execution by not using external and just call the function that was an 'external' and invoking it directly in my coroutine.   The scheduler is blocked and not proceeding either way.

## Use Case
In my design we are using a Versal Device from Xilinx.  The CIPS (i.e. processor) is provided by Xilinx using a SystemC model, TLM ports, and a socket to send and receive commands for something external to the simulator, i.e. QEMU, etc.  In my simulation environment, we have a COCOTB test that is expected to connect to the CIPS socket and send/receive commands from the CIPS model.  Since socket calls are blocking, I wrapped them in a function that is decorated by 'external'.  I then had a 'forked' coroutine to process incoming packets and trigger events.  With the current implementation in COCOTB however the simulator hangs because it is blocked in COCOTB waiting for the my socket read() to return before proceeding which will never happen because the simulator model is responsible for sending the packet.  If 'external' behaved the way it did back in v1.0, what I have implemented would work without any issues.

ktbarrett, 2021-12-14:
Likely changed in https://github.com/cocotb/cocotb/pull/565.

The implementation of external used to be implemented entirely within the the decorator; there was no scheduler interaction. You can probably copy/paste and see if this works for you.

https://github.com/cocotb/cocotb/blob/c1e259c1ba739d244ea2cdb4f857ea54e83d17a2/cocotb/decorators.py#L296-L322

We will have to discuss `external` and `function` with the maintainers and community to see if the current behavior is valuable to them, or not. I'm going to go out on a limb and guess almost no one uses this functionality and we can just change it back to the way it worked in 1.0 without peeving too many people off, but we will see.

eric-wieser, 2021-12-14:
As I understand it, the point of external is that it is _suppose_ to block the scheduler, up until it calls something marked with `@cocotb.function`, at which point the external thread is paused and the thread resumes.

ktbarrett, 2021-12-14:
If that is the intended use case (IMO that sounds like a problem exactly 1/2 of a person has ever had), then the documentation needs to be clarified. `external`/`function` are pretty bad names too, as the intent is to just interoperate threads with coroutines; but there's not much we can do about that now.

So we need a new interface. One specifically for this use case of calling a thread, looking like a `RunningTask`, where the thread may do stuff while the simulator is running, but is incapable of waking up the cocotb scheduler (i.e. calling `function`s in these thread is not allowed). The original implementation of `external` seems like a good starting off point for supporting that use case.

p0llard, 2022-02-21:
> We will have to discuss external and function with the maintainers and community to see if the current behaviour is valuable to them, or not. I'm going to go out on a limb and guess almost no one uses this functionality and we can just change it back to the way it worked in 1.0 without peeving too many people off, but we will see.

> As I understand it, the point of external is that it is supposed to block the scheduler, up until it calls something marked with @cocotb.function, at which point the external thread is paused and the thread resumes.
 
> If that is the intended use case (IMO that sounds like a problem exactly 1/2 of a person has ever had), then the documentation needs to be clarified. external/function are pretty bad names too, as [...]
 
Just to provide some clarification regarding the original intended use for `function`/`external`, it's particularly useful when you have a CPython extension that doesn't know about coroutines/asyncio: the test can `yield`/`await` an `external`-decorated function from an extension, which in turn reaches back into the interpreter and _calls_ a `function`-decorated cocotb couroutine, and everything should  _just work_; this is functionally equivalent to `yield`/`await`ing a normal coroutine (the extension function) that in turn `yield`/`await`s another coroutine (the `@function`-decorated coroutine that lives outside the extension).

The alternative is to write CPython extensions that are fully coroutine/asyncio-aware, but this way it's possible to take "real" code that normally executes against real hardware and turn it into a CPython extension that can be used for cosimulation with very little effort.

A concrete example, illustrated through register access:

Consider a C++ abstraction over a hardware register that looks something like the following (with extraneous details glossed over):

```c++
class Register {
    public:
        uint32_t read() { return impl->read(); };
        void write(uint32_t val) { return impl->write(val); };
 
    private:
        class RegisterImpl;
        RegisterImpl* impl;
};
```

This can be wrapped into a CPython extension such that a cosimulation `impl` is used that can reach into the interpreter and call `function`-decorated coroutines.
 
The methods `Register::read` and `Register::write` are exposed to the test through the wrapper and are decorated with `external`.
 
The test can then use the "real" register interface to manipulate registers in the hardware, with a call stack that looks roughly like:

```
yield cocotb.external(register.read)()

    --> // inside Register::read()
        return impl->read();

        --> // inside RegisterImpl::read()
            return sim.read(address);

            --> # inside cocotb interpreter
                @cocotb.function
                def read(address)
                    x = yield tb.read(address)
                    raise ReturnValue(x)
```

Higher level software abstractions over the hardware can also be wrapped into CPython extensions in the same manner, and the tests can use exactly the same software interfaces to interact with the hardware as real production code.

eric-wieser, 2022-02-21:
Thanks @p0llard, that's exactly the use case I had in mind too

likewise, 2022-11-07:
I have arrived here because the documentation of @function and @external seem to suggest a blocking function can be called that does not stop the simulator while being blocked.

In my case, I would like to concurrently feed an AXI Streaming source in the testbench to be fed with packets arriving at a Linux (TUN/)TAP interface, whilst also feed the TAP interface with packets arriving on an AXI Streaming sink in the test bench.

I would envision two background tasks that block on the arrival of packets, which call os.read() resp. tb.sink.recv() in blocking fashion, but generate simulator events once they unblock. However, I could not get even two tasks to run concurrently using @cocotb.function, @cocotb.external and @cocotb.start_soon().

ktbarrett, 2024-03-20:
> I have arrived here because the documentation of https://github.com/function and @external seem to suggest a blocking function can be called that does not stop the simulator while being blocked.

Based on what @p0llard mentioned, this is not the intended use case. The intended use case is imply to bridge the cocotb scheduler through code which is not `async`.

> I would like to concurrently feed an AXI Streaming source in the testbench to be fed with packets arriving at a Linux (TUN/)TAP interface

Unfortunately this is not possible. You cannot call VPI/VHPI/FLI functions from a totally external thread. You can only call such functions from a VPI callback. The only way to work around this is to make a little "server" task in cocotb which wakes up often (`Timer(10, 'ns')`, `RisingEdge(clk)`, whatever) and *then* poll the TAP interface for packets, sending them into the simulator via cocotb. 

Currently `cocotb.function` is allowed to be called from completely external threads, but this is actually *illegal*.