cocotb/cocotb #2812: Simulation running time

PiyushSaini09, 2021-12-20:
Hi-

I just started using cocotb (version currently used is 1.2) and was trying to figure few things. I got a few questions regarding cocotb-
1) How do I specify the time for which my simulation should run?

Right now I'm doing this using fork and join of  `Clock()` and the simulation ends for me as intended.
[image]

But I want to do this differently. I want to specify the time for which the simulation should run.
Earlier, I used to specify this time in a TCL script (for example: `run 10us`) which states my second query.

2) How do I make the TCL script my target in cocotb? Do I need to specify it in EXTRA_ARGS or is there another method?
I reference issue [#2754](https://github.com/cocotb/cocotb/issues/2754) but couldn't find any answer to this question.

imphil, 2021-12-20:
Hi, and welcome to cocotb!

First of all, please use the most recent stable version, that is 1.6.1 at the moment. We cannot really support version 1.2 any more, which is *really* old.

Can you let us know a bit more about your use case? Using `Timer()` to wait for a certain amount of time should do what you're looking for: let the simulation run for a certain amount of time, and then either end the simulation, or do a check (which you'd probably want to do?).

PiyushSaini09, 2021-12-20:
I'm actually using cadence's chamber for my work and I've asked them to update my cocotb version (I can't do it myself as I dont have such access). 
`Timer()` does actually fulfill my requirement. But what I want is to be able to provide the simulation run time by an external TCL script. Is it possible in cocotb to do such a thing (give a target to TCL script)? You can provide me an solution in reference to version 1.6.1. I'll move to it anyway.

P.S.: I'm really enjoying cocotb :)

stefan-muc, 2021-12-20:
Depends on if you know the simulation time beforehand or if you want to change it during simulation run.
In the end, a running simulation also "just" is a running Python script, so you can do everything you also can do in Python. The easy one would be to set an environment variable and then run simulation (and read environment variable there), a more advanced would be to communicate in some way, e.g. by reading a text file during simulation - this can also be changed during simulation run - or using some other kind of communication.

PiyushSaini09, 2021-12-20:
The way I'm using, I already know the simulation time beforehand. I thought of setting an environment variable for TCL script but in that case I don't know how to access the script via makefile as I couldn't find anything in the documentation regarding the same. 
My code sort of implements your suggestion of "Python Script" using `fork` and `join`. The simulation runs until 50 clock cycles and then finishes. But this is kind of vague and I want to define the simulation time as "run for 10ms" (or such). This will allow me to run the simulation only for that specific time and would remove the burden off of me of calculating the simulation time value each time my source changes.

ktbarrett, 2021-12-23:
The makefiles are primarily for our regressions and for simple projects. If your needs aren't being met by the makefiles as they stand, you are invited to modify them for your purpose or [do something else entirely](https://docs.cocotb.org/en/stable/custom_flows.html). We are not a project automation framework.

PiyushSaini09, 2021-12-27:
Understood. I'll close this issue now

closed 2021-12-27 by PiyushSaini09