cocotb/cocotb #2815: DUT doesn't change state with newer ghdl versions

distributed, 2021-12-23:
I use cocotb to simulate and interact with a VHDL DUT with `ghdl`. The DUT was behaving as expected until recently I upgraded `ghdl` and since then the DUT does not seem to change states ever. A reset signal is driven asserted-deasserted and a clock is present, yet I see no registers change value, neither on reset nor clock edges. 

I updated from a ~0.37 version of `ghdl` (May 2020) to master of a couple a days ago, after which the described problem appears. I decided to `git bisect` ghdl and believe I found the "offending" commit in `ghdl`:

```
commit ab2fd3d52f149efcc9cc66f0a0a5e378a1d63918 (HEAD, refs/bisect/bad)
Author: Tristan Gingold <tgingold@free.fr>
Date:   Sun Aug 2 09:26:44 2020 +0200

    vhdl: handle force/release statements in translate and grt.  For #1416
```

I am unsure where to describe this issue first, but in the end I decided to do it here. Between the described commit and `ghdl` master are over 1400 commits and more than a year, so I would be surprised if `ghdl` was fundamentally broken over that span. My suspicion would rather go to an unexpected interaction between a user, cocotb in this case, and `ghdl`. I hope that starting at cocotb we can try to work out why the whole system behaves as it does.

Standard information
----------------------------

cocotb version: 1.6.1. (via pip)
OS: Ubuntu 18.04 x86_64
python: 3.6.9, I highly suspect it's the system one from `/usr/bin/python3`
simulator: ghdl, see detailed description above

Log output 
---------------

### non-working with `ghdl` commit `ab2fd3d`

```
make -f Makefile results.xml
make[1]: Entering directory '/home/memi/FPGA/framegensim'
\
/home/memi/usr/bin/ghdl -i --std=08   --workdir=sim_build --work=work /home/memi/FPGA/framegensim/../libero_oriole_main/hdl/frame_generator_master.vhd /home/memi/FPGA/framegensim/pixel_mini.vhd /home/memi/FPGA/framegensim/mipi_imu_fifo.vhd && \
/home/memi/usr/bin/ghdl -m --std=08   --workdir=sim_build -Psim_build --work=work frame_generator_master
/home/memi/FPGA/framegensim/pixel_mini.vhd:5:1:warning: package "pkg_enum" was also defined in file "/home/memi/FPGA/framegensim/mipi_imu_fifo.vhd" [-Wlibrary]
/home/memi/FPGA/framegensim/pixel_mini.vhd:13:1:warning: package "pkg_scala2hdl" was also defined in file "/home/memi/FPGA/framegensim/mipi_imu_fifo.vhd" [-Wlibrary]
/home/memi/FPGA/framegensim/pixel_mini.vhd:379:1:warning: entity "streamfifo" was also defined in file "/home/memi/FPGA/framegensim/mipi_imu_fifo.vhd" [-Wlibrary]
/home/memi/FPGA/framegensim/mipi_imu_fifo.vhd:5:1:warning: package "pkg_enum" was also defined in file "/home/memi/FPGA/framegensim/pixel_mini.vhd" [-Wlibrary]
/home/memi/FPGA/framegensim/mipi_imu_fifo.vhd:13:1:warning: package "pkg_scala2hdl" was also defined in file "/home/memi/FPGA/framegensim/pixel_mini.vhd" [-Wlibrary]
/home/memi/FPGA/framegensim/mipi_imu_fifo.vhd:379:1:warning: entity "streamfifo" was also defined in file "/home/memi/FPGA/framegensim/pixel_mini.vhd" [-Wlibrary]
MODULE=pushimg TESTCASE= TOPLEVEL=frame_generator_master TOPLEVEL_LANG=vhdl \
 /home/memi/usr/bin/ghdl -r --std=08  --workdir=sim_build -Psim_build --work=work frame_generator_master --vpi=/home/memi/host_utils/python3/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --vcd=waveform.vcd  
loading VPI module '/home/memi/host_utils/python3/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:110  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/memi/host_utils/python3/bin/python
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     Running on GHDL version 1.0-dev (v0.37.0-911-gab2fd3d5) [Dunoon edition]
     0.00ns INFO     Running tests with cocotb v1.6.1 from /home/memi/host_utils/python3/lib/python3.6/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1640252481
     0.00ns WARNING  Pytest not found, assertion rewriting will not occur
     0.00ns INFO     Found test pushimg.test_pushimg
     0.00ns INFO     running test_pushimg (1/1)
vpi_get_str: unhandled property
vpi_get_str: unhandled property
vpi_get_str: unhandled property
vpi_get_str: unhandled property
/home/memi/FPGA/framegensim/pushimg.py:110: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  memb <= 0
/home/memi/FPGA/framegensim/pushimg.py:123: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.abort_in <= 0
/home/memi/FPGA/framegensim/pushimg.py:124: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.timestamp_in <= 1337
/home/memi/FPGA/framegensim/pushimg.py:125: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.id_in <= 42
/home/memi/FPGA/framegensim/pushimg.py:128: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.coeff_in <= ((2<<24) | (50 << 16) | (152 << 8))
/home/memi/FPGA/framegensim/pushimg.py:130: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.vsync <= 0
/home/memi/FPGA/framegensim/pushimg.py:131: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.hsync <= 0
/home/memi/FPGA/framegensim/pushimg.py:132: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
[... 89 lines cut ...]
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
/home/memi/FPGA/framegensim/pushimg.py:139: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  reset <= 1
/home/memi/FPGA/framegensim/pushimg.py:142: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.vsync <= 1
line 0
/home/memi/FPGA/framegensim/pushimg.py:88: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.hsync <= 1
/home/memi/FPGA/framegensim/pushimg.py:93: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.px_data0_in <= int(img.y[y, x])
/home/memi/FPGA/framegensim/pushimg.py:94: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.px_data1_in <= int(img.uv[y, x])
/home/memi/FPGA/framegensim/pushimg.py:97: DeprecationWarning: Setting values on handles using the ``handle <= value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
  dut.hsync <= 0
```

ktbarrett, 2021-12-23:
> DUT doesn't change state with newer ghdl versions
> vhdl: handle force/release statements in translate and grt

Hmmmmm.

The regression tests pass, including for force/release. You'll have to provide a minimal example we can run.

distributed, 2021-12-23:
Thanks for your fast reply. Do you know these cases when asking for help publicly immediately helps you find your own mistakes? This is one of them :D

Reducing the testcase I found the issue. In short, I use the following loop to set all inputs to zero at the beginning of the simulation:

```python
async def zeroAllInputs(dut):
    # Try to set all inputs to 0 in order to have as few 'U' as possible
    # in the DUT state.
    # We iterate over the names of the dut and check whether the associated
    # attribute is a cocotb signal.
    for x in dir(dut):
        memb = getattr(dut, x)
        if isinstance(memb, cocotb.handle.ModifiableObject):
            print(f"setting {x} to 0")
            memb.value = 0
```

Clearly it doesn't check whether a signal is an input or, as I found out, an internal signal. Not calling this function makes the simulation behave normally as before. I can see that setting outputs (and maybe internal signals) does not make sense, but I also do not follow how it should give rise to the observed behavior, i.e. the DUT just not doing anything in the testbench.

I provide complete code below for your reference.

For me two questions remain:
1. Is this the right way for the system to behave, to make the DUT "inert"? If no, I'd feel that a failure on writing output signals would be the right thing to do, but I'm no expert.
2. How can I set all input signals to some value without spelling out every single one of them?

The code
=======

`Makefile`
-------------

```make
TOPLEVEL_LANG = vhdl
VHDL_SOURCES = $(shell pwd)/frame_generator_master.vhd
TOPLEVEL = frame_generator_master
MODULE = pushimg
GHDL_ARGS=--std=08

SIM_ARGS ?= --vcd=waveform.vcd

include $(shell cocotb-config --makefiles)/Makefile.sim
```

`pushimg.py`
-------
```python
import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import ClockCycles

async def zeroAllInputs(dut):
    # Try to set all inputs to 0 in order to have as few 'U' as possible
    # in the DUT state.
    # We iterate over the names of the dut and check whether the associated
    # attribute is a cocotb signal.
    for x in dir(dut):
        memb = getattr(dut, x)
        if isinstance(memb, cocotb.handle.ModifiableObject):
            print(f"setting {x} to 0")
            memb.value = 0

@cocotb.test(timeout_time=40000, timeout_unit="us")
async def test_pushimg(dut):
    ##########################################
    # comment out to restore normal behavior #
    ##########################################
    await zeroAllInputs(dut)

    clock = Clock(dut.pxd_clk, 20, units="ns")  # Create a 10us period clock on port clk
    cocotb.fork(clock.start())  # Start the clock

    reset = dut.n_rst
    dut.vsync.value = 0

    reset.value = 0
    await ClockCycles(dut.pxd_clk, 5)
    reset.value = 1
    await ClockCycles(dut.pxd_clk, 10)

    dut.vsync.value = 1
    await ClockCycles(dut.pxd_clk, 10)

    # test will fail if ""inputs" zeroed"
    assert bool(dut.vsyncd.value), "vsyncd register not set"
```

`frame_generator_master.vhd`
--------

```vhdl
-------------------------------------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;
--use ieee.std_logic_arith.all;
use ieee.numeric_std.all;       -- for the signed, unsigned types and arithmetic ops

entity frame_generator_master is
port(
     PXD_CLK                   : in  std_logic;
     n_rst                     : in  std_logic;  
     vsync                     : in  std_logic
     );
end frame_generator_master;

architecture struct of frame_generator_master is
  signal vsyncd : std_logic;
begin

testprocess2: process(PXD_CLK, n_rst)
begin
    if(n_rst = '0') then
        vsyncd<= '0';
    elsif(PXD_CLK'event and PXD_CLK = '1') then
        vsyncd<= vsync;
	end if;
end process testprocess2;

end struct;
```

ktbarrett, 2021-12-23:
This is likely a GHDL bug, I have seen this aberrant behavior before with GHDL. Thank you for the code.

> 1. Is this the right way for the system to behave, to make the DUT "inert"? If no, I'd feel that a failure on writing output signals would be the right thing to do, but I'm no expert.

No you typically do not try to set every signal in the design. You set top-level inputs and reset the design. 

> 2. How can I set all input signals to some value without spelling out every single one of them?

The only way this is possible is if you could discern what's a port and what its direction is. That currently isn't possible (#695).

distributed, 2021-12-23:
Thanks again for the quick reply. I'll try to make a ghdl bug report out of this. I hope they will be able to reproduce with cocotb as I will struggle to provide a standalone VPI example.

Regarding #695, that'd be great news for those who are chasing the Us in their simulations :D

closed 2021-12-23 by distributed

ktbarrett, 2021-12-23:
[image]

This is the wave output of the sim. You can see the clock running, the reset being run, and the value going high, but the output doesn't change.