cocotb/cocotb #2817: How to open .coverage file

PiyushSaini09, 2021-12-27:
Hi.
I'm using cocotb 1.4 and xcelium as my simulator. I used COVERAGE option to get the HDL coverage for my RTL. After successful compilation, a **.coverage** file was created in my directory . However, I'm unable to find a software that will open the file to view the coverage. Is there a way I can view the coverage of my RTL?

I've attached the snippet of the created file herewith for reference -
[image]

The file says it is in "SQLite format".

themperek, 2021-12-28:
`.coverage` is for the python/test part. Is this what you are looking for? See (https://coverage.readthedocs.io/en/coverage-4.3.3/cmd.html)

For RTL every simulator does it differently and you have to look it up in the simulator manual.

PiyushSaini09, 2021-12-29:
Thank you. I was looking for RTL coverage. I could do that using `EXTRA_ARGS` in makefile for xcelium.
I'll close the issue.

closed 2021-12-29 by PiyushSaini09