cocotb/cocotb #2818: any arguments (COMPILE_ARGS, ...) in makefile are being ignored

Ali-Flt, 2021-12-29:
I'm using cocotb 1.6.1 on windows 10 (x64) with Anaconda (Python 3.9.7) + QuestaSim as simulator.

I'm trying to add `-svinputport=net` as an argument for the vlog command but adding it as COMPILE_ARGS, EXTRA_ARGS, PLUS_ARGS, SIM_ARGS, ... doesn't affect the operation when I issue `make` command. Why could this be?
Also when I modify the `Anaconda\Lib\site-packages\cocotb\share\makefiles\simulators\Makefile.questa` file the change doesn't affect my make command either. Where does the makefile get what it should do then? I assumed I was changing the right file.

Ali-Flt, 2021-12-29:
After a `make clean` command and re-running `make` it works.

closed 2021-12-29 by Ali-Flt

ktbarrett, 2021-12-29:
This is expected behavior. If you change a variable you will likely have to rebuild everything.