cocotb/cocotb #2819: Calling asynchronous function containing ”yield“ in synchronous function

justalittlenoob, 2021-12-29:
I encountered some problems while trying to make a synchronous function that calls asynchronous functions.(python 3.6.9, cocotb 1.4.0)

As shown in the following example code. The `read_cb` function will call `read` function(in `FakeDriver` class).

After running, I get the error

```none
yield self._fake_lock()
RuntimeError: Task got bad yield: <cocotb.decorators.RunningCoroutine object at 0x7f7fecdbfe10>
```

What I want is

```none 
init FakerDriver
locking... 
locking done
read...
addr: 0x01 
unlocking... 
unlocking done
read done
```

```python
import cocotb
import asyncio
from cocotb.decorators import coroutine
from cocotb.triggers import Event

class FakeDriver():
    def __init__(self):
        print("init FakeDriver")
        self.busy_event = Event("driver_busy")
        self.busy = False

    @coroutine
    def read(self, addr):
        print("read...")

        yield self._fake_lock()
        print("addr: ", addr)
        self._fake_unlock()

        print("read done")

    @coroutine
    def _fake_lock(self):
        print("locking...")
        if self.busy:
            yield self.busy_event.wait()

        self.busy_event.clear()
        self.busy = True
        print("locking done")

    def _fake_unlock(self):
        print("unlocking...")
        self.busy = False
        self.busy_event.set()
        print("unlocking done")

def read_cb():
    dri = FakeDriver()
    loop = asyncio.get_event_loop()
    task = loop.create_task(dri.read("0x01"))
    ret = loop.run_until_complete(task)
    loop.close()

if __name__ == "__main__":
    read_cb()

```

cmarqu, 2021-12-29:
I just answered in https://stackoverflow.com/a/70519512/41347 before seeing this.

justalittlenoob, 2021-12-31:
Problem solved according to your advice. thank you.

closed 2021-12-31 by justalittlenoob