cocotb/cocotb #2820: Difference Between async generators and Coroutines?

PiyushSaini09, 2022-01-03:
Hi-
I've just moved to COCOTB v1.6 from v1.2 and encountered async generators.
What exactly is the difference between **async generators** and **coroutines**?
I couldn't get the exact idea between the two when I referred documentation.

ktbarrett, 2022-01-03:
There are two valid way to write coroutine functions in cocotb: decorated generator functions and async functions. Decorated generator functions are the legacy way for writing coroutines and are deprecated; do not write new functions in this way.

**decorated generator function**
```python
@cocotb.coroutine
def legacy_coroutine():
    yield Timer(1, 'ns')
```

**async function**
```python
                            # notice no decorator
async def new_coroutine():  # notice the use of the keyword `async`
    await Timer(1, 'ns')    # notice change from `yield` to `await`
```

Python also has the concept of async generator functions, which are `async` functions that contain `yield`s. **These are not valid cocotb coroutines**. They cannot be used in `await` statements, nor can they be `fork`ed. You must use constructs like `async for` to iterate over an async generator.

PiyushSaini09, 2022-01-03:
Understood. Thank you.
I'll close the issue.

closed 2022-01-03 by PiyushSaini09