cocotb/cocotb #2821: Segmentation fault with CVC simulator and SDF annotation

jarvis-lorry, 2022-01-04:
I'm trying to run a simulation of the example adder with a synthesized version of the design that includes a Verilog PDK library file and an SDF annotation file.  CVC64 allows for SDF values to be annotated with a command line option but I cannot get the example simulation to run without a segmentation fault.

cocotb version: cocotb v1.6.1 
Operating system: x86_64-rhel6x
Simulator: OSS_CVC_7.00b-x86_64-rhel6x of 07/07/14 (Linux-elf).
Python: python3.8 run on system

Here is the error message I get: 

     0.00ns INFO     Running on CVC Verilog Simulator version OSS_CVC_7.00b-x86_64-rhel6x of 07/07/14 (Linux-elf).
     0.00ns INFO     Running tests with cocotb v1.6.1 from .../.local/lib/python3.8/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1641338601
     0.00ns WARNING  Pytest not found, assertion rewriting will not occur
     0.00ns INFO     Found test test_adder.adder_basic_test
     0.00ns INFO     Found test test_adder.adder_randomised_test
     0.00ns INFO     running adder_basic_test (1/2)
make[1]: *** [.../.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.cvc:75: results.xml] Segmentation fault (core dumped)
make[1]: Leaving directory '.../cocotb_system/cocotb/designs/adder_gate/tests'
make: *** [.../.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

themperek, 2022-01-05:
Looks like a CVS issue. You can try to run without cocotb first.

jarvis-lorry, 2022-01-15:
Hi, I've reworked the Verilog cell library so that it runs cleanly in the CVC simulator and I can compile the design without issue using just CVC but when I run a simulation in cocotb I still get a segmentation fault.  I've looked at this for awhile and can't see what the problem might be.  I can provide the verilog compilation log for CVC without cocotb to prove that there aren't any Verilog problems.  I've tried putting in +verbose and +libverbose options in CVC to see if that yields anything, I just can't figure out why cocotb won't run the sim.  

From cocotb:
Highest level modules:
adder
  Begin compiled design load:
  Begin simulation:

     0.00ns INFO     Running on CVC Verilog Simulator version OSS_CVC_7.00b-x86_64-rhel6x of 07/07/14 (Linux-elf).
     0.00ns INFO     Running tests with cocotb v1.6.1 from /home/lekan/.local/lib/python3.8/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1642206754
     0.00ns WARNING  Pytest not found, assertion rewriting will not occur
     0.00ns INFO     Found test test_adder.adder_basic_test
     0.00ns INFO     Found test test_adder.adder_randomised_test
     0.00ns INFO     running adder_basic_test (1/2)
make[1]: *** [/home/lekan/.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.cvc:75: results.xml] Segmentation fault (core dumped)
make[1]: Leaving directory '/home/lekan/cocotb_system/cocotb/designs/adder_gate/tests'
make: *** [/home/lekan/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

marlonjames, 2022-01-15:
Can you run with `COCOTB_LOG_LEVEL=TRACE` and `COCOTB_SCHEDULER_DEBUG=1` and capture the output?

jarvis-lorry, 2022-01-17:
Hi, I've run with settings you suggested and here is the output:

make -f Makefile results.xml
make[1]: Entering directory '/home/lekan/cocotb_system/cocotb/designs/adder_gate/tests'
mkdir -p sim_build
MODULE=test_adder \
TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=verilog \
        /home/lekan/open-src-cvc/bin/cvc64  +acc+2 -o sim_build/sim.vvp +define+COCOTB_SIM=1 +loadvpi=/home/lekan/.local/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.so:vlog_startup_routines_bootstrap -sv +define+UNIT_DELAY="" -v /home/lekan/OpenLane/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/sky130_fd_sc_hd_primitives.v +fstvars -v /home/lekan/OpenLane/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/sky130_fd_sc_hd_cells_nospecify.v  ../adder.resized.v
OSS_CVC_7.00b-x86_64-rhel6x of 07/07/14 (Linux-elf).
Copyright (c) 1991-2014 Tachyon Design Automation Corp.
  All Rights reserved.  Licensed software subject to prohibitions and
  restrictions.  See OSS CVC artistic license included with release.
Today is Mon Jan 17 14:33:36 2022.
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
Elaborating source file "../adder.resized.v"
**../adder.resized.v(218) WARN** [531] TAP_36(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(220) WARN** [531] TAP_37(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(222) WARN** [531] TAP_38(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(224) WARN** [531] TAP_39(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(226) WARN** [531] TAP_40(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(228) WARN** [531] TAP_41(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(230) WARN** [531] TAP_42(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(232) WARN** [531] TAP_43(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(234) WARN** [531] TAP_44(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(236) WARN** [531] TAP_45(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(238) WARN** [531] TAP_46(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(240) WARN** [531] TAP_47(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(242) WARN** [531] TAP_48(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(244) WARN** [531] TAP_49(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(246) WARN** [531] TAP_50(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(248) WARN** [531] TAP_51(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(250) WARN** [531] TAP_52(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(252) WARN** [531] TAP_53(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(254) WARN** [531] TAP_54(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(256) WARN** [531] TAP_55(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(258) WARN** [531] TAP_56(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(260) WARN** [531] TAP_57(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(262) WARN** [531] TAP_58(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(264) WARN** [531] TAP_59(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(266) WARN** [531] TAP_60(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(268) WARN** [531] TAP_61(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(270) WARN** [531] TAP_62(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
**../adder.resized.v(272) WARN** [531] TAP_63(sky130_fd_sc_hd__tapvpwrvgnd_1) explicit connection list fewer ports 2 connected than type's 4
[... cut ...]
     0.00ns DEBUG    Scheduled coroutine Scheduler._do_writes
     0.00ns DEBUG    All coroutines scheduled, handing control back to simulator
     0.00ns TRACE    Returning to simulator
make[1]: *** [/home/lekan/.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.cvc:75: results.xml] Segmentation fault (core dumped)
make[1]: Leaving directory '/home/lekan/cocotb_system/cocotb/designs/adder_gate/tests'
make: *** [/home/lekan/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

jarvis-lorry, 2022-01-17:
I have question about Makefile.cvc in .../.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.cvc

The comment says that only interpreted mode works for CVC; however in Makefile.cvc, the simulation looks for an executable, sim.vvp.  I don't think there is an executable created in interpreted mode.

#only interpreted mode works for the moment
CVC_ITERP ?= 1

jarvis-lorry, 2022-01-17:
I unintentionally closed this issue.  Please see the debug information I posted above.

jarvis-lorry, 2022-01-17:
A couple more points.  I can run this same simulation using Icarus Verilog but I would prefer to use CVC because CVC allows for SDF annotation.  I can also run a different simulation with CVC and it works as well so it is possible to run a CVC with SDF annotation in cocotb.

marlonjames, 2022-01-19:
There is nothing obvious in the log. The test registers a 1ns `Timer` callback and a `ReadWrite` callback, then after returning to the simulator the segfault occurs.

jarvis-lorry, 2022-01-19:
After some trial and error, I've found that I can run the CVC simulation as long as I make sure to add +dumpvars as one of CVC command switches.  I don't know how this prevents the segmentation fault but it seems to work reliably.  cocotb still gives some errors but the simulation runs and VCD waveforms are dumped.

ktbarrett, 2022-01-25:
Sounds like the problem is solved? I'll close. Reopen if you there is still some issue.

closed 2022-01-25 by ktbarrett

avgsurfman, 2026-02-05:
I don't think it's an issue with the CVC. I wasted my day wrangling GNU dbg and I got this SEGFAULT:
```
Program received signal SIGSEGV, Segmentation fault.
0x0000000000586154 in do_sim () at v_sim.c:717
717        __cur_te_endp->tenxtp = __tefreelstp;
(gdb) bt
#0  0x0000000000586154 in do_sim () at v_sim.c:717
#1  0x0000000000585f77 in __pv_sim () at v_sim.c:624
#2  0x00000000004336bc in __dig_main (argc=9, argv=0x7ffd50969c68) at cvc.c:2051
#3  0x0000000000432aba in main (argc=9, argv=0x7ffd50969c68) at dig_main.c:176
(gdb) __cur_te_endp
Undefined command: "__cur_te_endp".  Try "help".
(gdb) print __cur_te_endp
$1 = (struct tev_t *) 0x0
(gdb) print __cur_te_hdrp
$2 = (struct tev_t *) 0x0
(gdb) frame 0
```

I think it's failing to interface with Cocotb through vpi, maybe?
```
     0.00ns INFO     cocotb.initialize                  Seeding Python random module with 1770314144
     0.00ns WARNING  gpi                                vpi_iterate returned NULL for type vpiInstance for object NULL
     0.00ns INFO     cocotb.initialize                  Initialized cocotb v2.1.0.dev0+9f986c9f from /home/de_generic/cocotb/src/cocotb
```

The time queue seems to be completely empty, which supports this:
```
[...]
 if (__compiled_sim) do_compiled_sim();
 else do_sim();
}

/*
 * run the actual simulation until all events are done
 */
static void do_sim(void)
{
 struct tev_t *tevp;

 tevp = __cur_te_hdrp;
 for (;;)
  {
   /* execute events until current time event list empty */
   /* events never added here (maybe to pound 0) */
   for (; tevp != NULL; tevp = tevp->tenxtp)
```

avgsurfman, 2026-02-19:
Any idea on how to fix this on cocotb side? @ktbarrett I'm honestly motivated enough to patch it myself but I don't know where to start. Is there an option in cocotb to revert to an older vpi interface (1995) with less options?