cocotb/cocotb #2823: Run tests with a substring

PiyushSaini09, 2022-01-05:
Hi -

Can I run tests with just a substring instead of writing the whole test name?
For example: 
I have the following five tests:
`test_greater,
 test_lesser,
 test_greatless,
 test_equal,
 test_notequal`
 
Is it possible that if I write the substring "less", cocotb runs all those tests that contain the string "less" (just like the way Pytest does since cocotb uses pytest to some extent) instead of writing the whole test case name? Right now, I am writing the complete test case name as `TESTCASE=test_lesser` to run. Am I doing something wrong in this or I missed something?

P.S.: I'm using cocotb version 1.6.0 and xcelium as my simulator.

themperek, 2022-01-05:
I do not think we have such functionality. 

One could add such matching here with [fnmatch](https://docs.python.org/3/library/fnmatch.html) or [re](https://docs.python.org/3/library/re.html): https://github.com/cocotb/cocotb/blob/cb876412fdaee249407619c1d4e34d3c04241754/cocotb/regression.py#L169

PiyushSaini09, 2022-01-05:
Understood. So I'll have to make changes in my copy of `regression.py` to get the required functionality?

themperek, 2022-01-05:
You can try. 

Actually, the right place is here I think:
https://github.com/cocotb/cocotb/blob/cb876412fdaee249407619c1d4e34d3c04241754/cocotb/regression.py#L183-L210

You can provide a PR also.

PiyushSaini09, 2022-01-05:
Thanks for the help. I'll close the issue for now. Will provide PR if I'm able to make the required change.

closed 2022-01-05 by PiyushSaini09