cocotb/cocotb #2826: Tox build failing on MacOS with Python 3.9

raysalemi, 2022-01-07:
Hi,

I'm trying to add py39 to a tox.ini file similar to the cocotb one, but I can't seem to install cocotb in the Python3.9 venv:

```
    File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1739, in _execute_child
      env_list.append(k + b'=' + os.fsencode(v))
    File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/os.py", line 810, in fsencode
      filename = fspath(filename)  # Does type-checking of `filename`.
  TypeError: expected str, bytes or os.PathLike object, not int
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 21.0.1; however, version 21.3.1 is available.
You should consider upgrading via the '/Users/raysalemi/repos/python4uvm_examples/.tox/py39-macos/bin/python -m pip install --upgrade pip' command.
```

This is a long dump after the `clang` command. 

Suggestions?

It works fine with Python 3.8.

marlonjames, 2022-01-07:
It is complaining about some value in the environment variables being an int instead of a string.

raysalemi, 2022-01-07:
> It is complaining about some value in the environment variables being an int instead of a string.

Why wouldn't that be a problem in 3.8?

raysalemi, 2022-01-07:
This works fine on GitHub, so it is a local installation issue.

closed 2022-01-07 by raysalemi