cocotb/cocotb #2829: How to keep questa session running when simulation is done

filamoon, 2022-01-10:
WSL2 ubuntu
cocotb 1.6.1
questa linux 2021.2_1 64b

I'm using cocotb/cocotb-test and questa cosimulation. Each time the simulation is done, questa(vsim) quits.
Is there any way to keep the questa session running after the simulation is done, so I can inspect the waves?
I have tried adding "-onfinish ask" to the vsim CLI but it did not help.
It takes an incredibly long time to launch a vsim session.

Thanks.

ktbarrett, 2022-01-10:
This seems like a question for cocotb-test. Pinging @themperek.

themperek, 2022-01-10:
You can use "WAVES=1" (https://github.com/themperek/cocotb-test#environmental-variables) as an environmental variable or use `waves` parameter got `run` function (https://github.com/themperek/cocotb-test#arguments-for-simulatorrun).

Please use https://github.com/themperek/cocotb-test/issues for `cocotb/cocotb-test` related questions.

themperek, 2022-01-10:
Correction: to not quit use `gui` parameter (https://github.com/themperek/cocotb-test#arguments-for-simulatorrun).

jonpovey, 2022-01-25:
Commented on the cocotb-test issue (the feature is working for me)
Suggest closing this cocotb ticket.

closed 2022-01-25 by ktbarrett