cocotb/cocotb #2830: Should Makefile and cocotb testbench be in same directory?

PiyushSaini09, 2022-01-12:
Hi-

Is it necessary to keep the `Makefile` and my cocotb testbench in the same directory in order to run?

I can provide the path to the verilog source but I couldn't do so for the python source. I even tried giving path to the module(which is the python file) but it obviously didn't work because module expected just the name of the file. 
[image]

Also, when I run the `make` command, some files are created by xcelium and cocotb into the current working directory. What I want to do is keep makefile in directory 1(assume name as dir1) and python testbench in another(assume name as dir2). So, now after I run `make` I need those created files in dir1 and nothing should go in dir2. Is it possible to do this?

Thanks.

P.S.: I'm using cocotb v1.6.0

themperek, 2022-01-14:
Not sure if this answer your question but:

You can use `PYTHONPATH` to specify python source files.
https://github.com/cocotb/cocotb/blob/1536fd14dddaed797ad5d68858baad1a5f578076/examples/adder/tests/Makefile#L34

PiyushSaini09, 2022-01-18:
Yes. It works as expected. Thank you for the help :)
I'll close the issue.

closed 2022-01-18 by PiyushSaini09