cocotb/cocotb #2831: Suggest deletion of ancient branches

jonpovey, 2022-01-21:
I notice there are some very old branches in the repo that look like they could be deleted to reduce clutter, e.g. the ones >5 years old.

ktbarrett, 2022-01-22:
* moved `wishbone` to cocotb-bus
* deleted `interfaces` as it's OBE
* deleting 2022-01-24 `Issue#23` since the issue was closed as out-of-scope
* deleting 2022-01-24 `Issue#134` since it's a pretty lame solution to the problem.

jonpovey, 2022-01-24:
Thanks. I don't see the Issue branches deleted at this time.
```
Pruning origin
URL: https://github.com/cocotb/cocotb
 * [pruned] origin/cmarqu-issue-template
 * [pruned] origin/doc-add-mixedlang-for-xcelium
 * [pruned] origin/interfaces
 * [pruned] origin/windows-tox-conda
 * [pruned] origin/wishbone
```

closed 2022-01-25 by ktbarrett