cocotb/cocotb #2833: Request to support data real for SV-RNM

mmatzev, 2022-01-21:
I tried to run a SV-RNM with verilator and cocotb  version 1.6.1.
It gave me a error message that the input data type float is not supported. It would be nice to support this.

A quick and dirty hack to a elif statement for float ended with an error about a exception in a coroutine.I don't know the internals of cocotb and do not understand how the value handling is really done. Therefore, I can't provide a patch. Sorry. 

Quick & dirty add in to handle.py:
[image]

Error message on debian WSL1:
[image]

cmarqu, 2022-01-21:
Just quickly, https://github.com/cocotb/cocotb/pull/1282 says it should work. Might be a Verilator thing?

ktbarrett, 2022-01-21:
Verilator has poor support for reals IIRC.

closed 2022-01-21 by ktbarrett

ktbarrett, 2022-01-24:
Yes cocotb complains, but it complains because it thinks you are trying to set a real value on a logic or array of logic (ModifiableObject). This is due to Verilator reporting the actual type of the signal as `vpiReg` rather than a more descriptive type.

The patch works because the signal is actually a real under the hood, so the VPI call doesn't fail. However, allowing users to set reals on ModifiableObject in general may allow them to unsafely set reals on signals that aren't actually reals. This may fail politely or crash hard, depending on the simulator. We don't want to enable that kind of failure. It is better for Verilator to emit more expressive VPI types on signals.