cocotb/cocotb #2836: Running questa with VHPI not possible without FLI

imikkl, 2022-01-24:
Using latest relase of cocotb 1.6.1 to date.
Trying to run a project with SIM=questa and VHDL_GPI_INTERFACE=vhpi results in the error
 
ERROR: cocotb was not installed with an FLI library, as the mti.h header could not be located.
If you installed an FLI-capable simulator after cocotb, you will need to reinstall cocotb.

even though FLI is not used. This seems to come from a wrong check for FLI in Makefile.questa. Moving the check for FLI into the FLI-only section fixes this:
[Makefile-questa-vhpi.patch.txt](https://github.com/cocotb/cocotb/files/7927258/Makefile-questa-vhpi.patch.txt)

ktbarrett, 2022-01-24:
Good catch. I'll incorporate it. Just a heads up, Questa's VHPI and cocotb don't work effectively together until Questa 2022.1.

imikkl, 2022-01-25:
Thanks for the info. I got some more or less random (depending on test bench code) backtraces having to do with TriggerException and vhpiCbAfterDelay and thought these might be the problems having to do with the experimental VHPI status so maybe they will fix themselves in the future. They do no appear using FLI.

ktbarrett, 2022-01-25:
The issue you are seeing is #2749.

PR #2838 (2022-01-27): Fix some Questa issues
Closes #2836. Closes #2837. Pinging @imikkl. If you give me a name and email I can make you author on the first commit.
closed 2022-01-27 by ktbarrett