cocotb/cocotb #2837: Installing cocotb with FLI for Questa-2021.2 not possible: vdbg obsolete

imikkl, 2022-01-24:
Using latest relase of cocotb 1.6.1.
After installing cocotb and when trying to run a project with SIM=questa it results in the error:

ERROR: cocotb was not installed with an FLI library, as the mti.h header could not be located.
If you installed an FLI-capable simulator after cocotb, you will need to reinstall cocotb.

However, questa is installed and vdbg executable is in PATH but invoking vdbg results in:

vdbg command is now obsolete. Please run vopt with "-debugdb" option to enable schematic and causality fuctionalities.

Is there a way to fix this?

ktbarrett, 2022-01-24:
This is kind of a sore point. cocotb must be built with vdbg in your path for cocotb to build support for the FLI. Once cocotb is built, pip stores the built object in a cache, so any successive attempt to install the project will use the cached version, never allowing the package to be rebuilt with FLI support. You will have to add the `--no-cache-dir` when using `pip install` to override pip's wheel cache. 

As for vdbg being deprecated, we could move to another executable (say `vsim`).

themperek, 2022-01-24:
>As for vdbg being deprecated, we could move to another executable (say vsim).

I think there is also `vsim` in Riviera. At the time I tied something that is only in Questa. 
Need to find something else.

ktbarrett, 2022-01-24:
How about detect vsim and then see if mti.h is present? I would hope Riviera doesn't ship a file called mti.h that is incompatible.

imikkl, 2022-01-25:
> This is kind of a sore point. cocotb must be built with vdbg in your path for cocotb to build support for the FLI. Once cocotb is built, pip stores the built object in a cache, so any successive [...]

Thanks, '--no-cache-dir' was the missing thing and it is working now. I did not find this in the installation docs.

> As for vdbg being deprecated, we could move to another executable (say `vsim`).

I see. So the functionality of vdbg is actually not needed but only its existence is used for checking for FLI support? Maybe then vdbg is not going anywhere despite it being obsolete. All the 'executables' in '<questa_install>/bin/' are a symlink to '<questa_install>/vco' anyway.

ktbarrett, 2022-01-25:
>Thanks, '--no-cache-dir' was the missing thing and it is working now. I did not find this in the installation docs.

Because it's a corner case. Perhaps it's worth a blurb.

closed 2022-01-27 by ktbarrett

mvsoliveira, 2022-02-26:
Hi,
I encountered the same issue but `sudo pip3 --no-cache-dir install cocotb` did not work for me :(
The installation finishes without any error. 
I am using Questa 2020.4 in a Centos 7 OS.
These are the generated libraries:

-rwxr-xr-x. 1 root root  35056 Feb 26 17:29 libcocotb.so
-rwxr-xr-x. 1 root root  10576 Feb 26 17:29 libcocotbutils.so
-rwxr-xr-x. 1 root root 386816 Feb 26 17:29 libcocotbvhpi_aldec.so
-rwxr-xr-x. 1 root root 385584 Feb 26 17:29 libcocotbvhpi_ius.so
-rwxr-xr-x. 1 root root 386816 Feb 26 17:29 libcocotbvhpi_modelsim.so
-rwxr-xr-x. 1 root root 358384 Feb 26 17:29 libcocotbvpi_aldec.so
-rwxr-xr-x. 1 root root 358384 Feb 26 17:29 libcocotbvpi_ghdl.so
-rwxr-xr-x. 1 root root 362984 Feb 26 17:29 libcocotbvpi_icarus.vpl
-rwxr-xr-x. 1 root root 358384 Feb 26 17:29 libcocotbvpi_ius.so
-rwxr-xr-x. 1 root root 358672 Feb 26 17:29 libcocotbvpi_modelsim.so
-rwxr-xr-x. 1 root root 358384 Feb 26 17:29 libcocotbvpi_vcs.so
-rwxr-xr-x. 1 root root 358384 Feb 26 17:29 libcocotbvpi_verilator.so
-rwxr-xr-x. 1 root root  10968 Feb 26 17:29 libembed.so
-rwxr-xr-x. 1 root root  19000 Feb 26 17:29 libgpilog.so
-rwxr-xr-x. 1 root root 166320 Feb 26 17:29 libgpi.so
-rwxr-xr-x. 1 root root  23840 Feb 26 17:29 libpygpilog.so

vdbg is in the path and mti.h exists:

`which vdbg`
`/eda/mentor/2020-21/RHELx86/QUESTA-CORE-PRIME_2020.4/questasim/linux_x86_64/vdbg`

`ll /eda/mentor/2020-21/RHELx86/QUESTA-CORE-PRIME_2020.4/questasim/include/mti.h`
`-r--r--r--. 1 500 500 33074 Oct 14  2020 /eda/mentor/2020-21/RHELx86/QUESTA-CORE-PRIME_2020.4/questasim/include/mti.h`

`python3 --version`
`Python 3.6.8`

Could you please give me a direction?

ktbarrett, 2022-02-26:
@mvsoliveira Did you uninstall cocotb first, or add the `--force-reinstall` option? by default pip will not reinstall if version conditions are met.

mvsoliveira, 2022-02-26:
Dear @ktbarrett,

Thanks for the quick reaction. I uninstalled it and did not help :( 
I tried to use --force-reinstall but it only has any effect when upgrading the package version apparently.

ktbarrett, 2022-02-26:
Can you run `pip install -v --force-reinstall --no-cache-dir cocotb` and post the output here? With verbose mode you should be getting messages about why it is deciding to not build the FLI.

mvsoliveira, 2022-02-26:
Based on your directions, I found the solution.

I was running pip with sudo, and vdbg did not exist in the path for the root user. 
I added the questasim tools to the root path, uninstalled, and installed cocob again with the no-cache-dir option and it fixed the issue. 
Thanks a lot!!!

ktbarrett, 2022-02-26:
@mvsoliveira I would advise against installing packages with sudo... Most systems have a user install directory. you can use `pip install -U ...` to install to the user directory. Newer version of `pip` already do that for you.

PR #2838 (2022-01-27): Fix some Questa issues
Closes #2836. Closes #2837. Pinging @imikkl. If you give me a name and email I can make you author on the first commit.