cocotb/cocotb #2839: factory ise when changing form 1.5.2 to 1.6.1

bth105, 2022-01-26:
Hi I have just updated from 1.5.2 to 1.6.1 and Im getting a error from my run_test. I generate my tests using the TestFactory, this passes in the name of a function into the data_in input on run_test. But when I do `for transaction in data_in():` I get the error 
TypeError: 'NoneType' object is not callable
There must be a change in the use case between 1.5.2 and 1.6.1 but i cant find any documentation on it.

bth105, 2022-01-26:
@cocotb.coroutine
def run_test(dut, data_in=None, idle_inserter=None, backpressure_inserter=None):

    cocotb.fork(Clock(dut.clk, 5000).start())
    tb = EthernetStripperTB(dut)

    if backpressure_inserter is not None:
        tb.backpressure.start(backpressure_inserter())

    yield tb.reset()
    dut.dout_ready = 1

    # Send in the packets
    for transaction in data_in():
        yield tb.input.send(raw(Ether()/transaction))

    yield Timer(1000000)
    yield RisingEdge(dut.clk)

    raise tb.scoreboard.result

def random_packet_sizes(min_size=64, max_size=1024, npackets=100):
    """random string data of a random length"""
    for i in range(npackets):
        yield get_bytes(random.randint(min_size, max_size), random_data())

factory = TestFactory(run_test)
factory.add_option("data_in", [random_packet_sizes])
factory.add_option("backpressure_inserter", [None, intermittent_single_cycles])
factory.generate_tests()

ktbarrett, 2022-01-26:
Looks like support for `@cocotb.coroutine` was accidentally broken. We missed this because we updated all the tests to use async/await.

PR #2841 (2022-01-31): Fix TestFactory to work with legacy-style tests
Wrapping a wrapped function-like is breaking something. Going to not do
that... Turns out it wasn't even needed. It should have been removed
with the other testfactory fix in v1.6.1.

Cherry-pick of #2840. Closes #2839.
closed 2022-02-01 by ktbarrett