cocotb/cocotb #2842: Unable to install cocotb on MSYS2 environment

Ogekuri, 2022-01-28:
I'm tring to install the cocotb on MSYS2 environment but it fail on binaries build on missing 'unistd.h'.

```
"C:/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/HostX86/x64/cl.exe" /c /nologo /O2 /GL /DNDEBUG /MD /W4 -DPYGPILOG_EXPORTS= -D__STDC_FORMAT_MACROS= -DWIN32= -Icocotb/share/include -IC:/msys64/mingw64/include -IC:/msys64/mingw64/include/python3.9 "-IC:\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.7.2\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tpcocotb/share/lib/py_gpi_log/py_gpi_logging.cpp /Fobuild/temp.win-amd64-3.9/Release/cocotb/libs/libpygpilog/cocotb/share/lib/py_gpi_log/py_gpi_logging.obj /permissive-
  py_gpi_logging.cpp
  C:/msys64/mingw64/include/python3.9\Python.h(36): fatal error C1083: Non è possibile aprire il file inclusione: 'unistd.h': No such file or directory

```
`**C:/msys64/mingw64/include/python3.9\Python.h(36): fatal error C1083: Non è possibile aprire il file inclusione: 'unistd.h': No such file or directory**
`
It seem be undefined MS_WINDOWS that is used to skip the 'unistd.h':
```
#ifndef MS_WINDOWS
#include <unistd.h>
#endif
```

I have follow the guide from [Tier 2 Setup Instructions](https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions)

Do you have any suggestion?

Thanks in advance
Francesco.

themperek, 2022-01-28:
What version of cocotb you are using? It may have been fixed: https://github.com/cocotb/cocotb/pull/1795

On MSYS you probably want to use GCC not MSVC to build libraries.

Ogekuri, 2022-01-28:
> What version of cocotb you are using?

I'm using current version on master branch but the issue still.

I've fixed the issue on **'unistd.h'** with an "`_extra_cxx_compile_args_msvc += ["-DMS_WINDOWS"]`" on **cocotb_build_libs.py** on the line 31, but now I'm stuck again on '**ieeefp.h**':
`C:\msys64\mingw64\include\python3.9\**pyport.h**(217): fatal error C1083: Non è possibile aprire il file inclusione: '**ieeefp.h**': No such file or directory`

> On MSYS you probably want to use GCC not MSVC to build libraries.

Yes, I think too, it was my first approch ... i've already tryed to force gcc with a **distutils.cfg**:
```
[build]
compiler=mingw32
[build_ext]
compiler=mingw32
```

But it fail, and unfortunatelly i can't undersood why:   **TypeError: 'NoneType' object is not iterable**

```
gcc -mdll -O -Wall -DGPILOG_EXPORTS= -D__STDC_FORMAT_MACROS= -DWIN32= -Icocotb/share/include -IC:/msys64/mingw64/include -IC:/msys64/mingw64/include/python3.9 -c cocotb/share/lib/gpi_log/gpi_logging.cpp -o build/temp.win-amd64-3.9/Release/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual  windres -i libgpilog.rc -o build/temp.win-amd64-3.9/Release/cocotb/libs/libgpilog/libgpilog.rc.o
  Traceback (most recent call last):
    File "C:\msys64\mingw64\lib\python3.9\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 363, in <module>
      main()
    File "C:\msys64\mingw64\lib\python3.9\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "C:\msys64\mingw64\lib\python3.9\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 261, in build_wheel
      return _build_backend().build_wheel(wheel_directory, config_settings,
    
....

      File "C:/Users/frolando.DLC0YLZW2/AppData/Local/Temp/pip-build-env-m4go02t8/overlay/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 550, in build_extension
      self.compiler.link_shared_object(
    File "C:/Users/frolando.DLC0YLZW2/AppData/Local/Temp/pip-build-env-m4go02t8/overlay/lib/python3.9/site-packages/setuptools/_distutils/ccompiler.py", line 713, in link_shared_object
      self.link(CCompiler.SHARED_OBJECT, objects,
    File "C:/Users/frolando.DLC0YLZW2/AppData/Local/Temp/pip-build-env-m4go02t8/overlay/lib/python3.9/site-packages/setuptools/_distutils/cygwinccompiler.py", line 181, in link
      libraries.extend(self.dll_libraries)
  TypeError: 'NoneType' object is not iterable
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
```

ktbarrett, 2022-01-28:
Just tried installing cocotb using a fresh msys2 install. It doesn't give the above error, but it does fail to build. Looks like a bug in their GCC. But our CI works...

I no longer have *any* trust in msys2. I'm going to start recommending people to not use it for cocotb.

@Ogekuri 
If you are comfortable with msys2, you will likely find WSL comfortable as well. I would use that instead.

Ogekuri, 2022-01-28:
> If you are comfortable with msys2, you will likely find WSL comfortable as well. I would use that instead.

Sure, I know. As well I can use just linux, but anyway it's not the point.

I've cocotb running on colab (minicolab3) that work fine with iverilog but it's very difficult to get ghdl or verilator work properly on that enviroment.

On the other side, I've MSYS where iverilog ghdl ghdl-gcc and verilator work fine with a very smooth install, but not the cocotb.

Ihmo make sense to support MSYS on windows, that provide more simulator "ready to run".

themperek, 2022-01-28:
Did you enter msys2 with `mingw32/64.exe`?

ktbarrett, 2022-01-28:
@Ogekuri hdl/containers was supposed to add cocotb as a supported package, not sure where that ended up. I would create an issue there to see if Unai can get something working for cocotb. msys2's Python environment is more than a little broken, so distributing it via pacman might be the sanest choice. However, this repo is not going to take that scope on.

Ogekuri, 2022-01-28:
> Did you enter msys2 with `mingw32/64.exe`?

I don't have any `64.exe `on my (fresh) installation on MSYS2, I've a `C:\msys64\mingw64.exe`, is it the same?

Ogekuri, 2022-01-28:
> so distributing it via pacman might be the sanest choice

Sure

ktbarrett, 2022-01-28:
https://github.com/hdl/MINGW-packages/issues/3

themperek, 2022-01-29:
As of now, this seems to work for me:
```bash
pacman -S --needed  mingw-w64-x86_64-python3-wheel mingw-w64-x86_64-python3-setuptools
pip3 install  --global-option build_ext --global-option --compiler=mingw32 --no-build-isolation https://github.com/cocotb/cocotb/archive/refs/heads/master.zip
```
¯\_(ツ)_/¯

Looks like something with setuptools. All those msys ports seem unstable.

themperek, 2022-01-29:
Seems like pip is trying to install the latest version that breaks compatibility.
```
setuptools-60.5.0-py3-none-any.whl 
wheel-0.37.1-py2.py3-none-any.whl 
```

themperek, 2022-01-29:
I have updated the wiki.

@Ogekuri Can you confirm? Close the issue in case this works

Ogekuri, 2022-01-29:
> @Ogekuri Can you confirm? Close the issue in case this works

I'm going to try, I'll start with a fresh instalation of MSYS, I've done to much changes on the current one .. I'll provide an update later...

Ogekuri, 2022-01-29:
> @Ogekuri Can you confirm? Close the issue in case this works

Yes, I can confirn that works!
Everything has worked smoothly, without errors or warning!
Sure, you can close the issue. 
**Your support was very appreciated, many many thanks!** :-)

Do you suggest me to use the pip version or I can use the master/branch version?

--

I recap here all steps I done to get cocotb working on MSYS2 for whoever will read the post in future:

0) Download the installer: `msys2-x86_64-20220128.exe` and install MSYS2 on `C:\msys64`
1) Execute a shell from `C:\msys64\mingw64.exe`
2) Execute a system update with: `pacman -Syu`
3) Install the packages needed: `pacman -S --needed make mingw-w64-x86_64-gcc mingw-w64-x86_64-python3-pip mingw-w64-x86_64-python3-setuptools mingw-w64-x86_64-python3-wheel` (40 packages was installed)
4) Install cocotb from master with: `pip3 install  --global-option build_ext --global-option --compiler=mingw32 --no-build-isolation https://github.com/cocotb/cocotb/archive/refs/heads/master.zip`
5) Install **iverilog**, **ghdl** and **verilator** with: `pacman -S --needed mingw-w64-x86_64-iverilog mingw-w64-x86_64-ghdl mingw-w64-x86_64-verilator` (7 packages was installed)
6) test the adder from example (verilog):
[image]

7) test the adder from example (vhdl):
[image]

closed 2022-01-29 by themperek

themperek, 2022-01-29:
>Do you suggest me to use the pip version or I can use the master/branch version?

Both should be OK.

ktbarrett, 2022-01-29:
Okay, so I see you have to use the mingw pip, setuptools, wheel, and gcc to build the package, the msys packages *don't* work (at least right now). I replicated msys environment failure a few times and even saw a few other people seeing the same issue I was having with gcc not being able to link because it can't find libraries in directories which are clearly in the user supplied search paths. And you have to use `--no-build-isolation` because the isolated build will fetch upstream versions of setuptools and wheel which will break. And all of this could break if they decide to change to non-standard behavior (https://github.com/msys2/setup-msys2/issues/167). What a nightmare.

omkars3, 2022-11-20:
Does anyone know how to install modelsim inside ming64?