cocotb/cocotb #2843: Dropping support for mingw environments and compiler

ktbarrett, 2022-01-28:
This includes dropping official support and installation instructions for anaconda + mingw and the msys2 and mingw32/64 environment. 

## Rationale
These environments have been broken a lot recently. We have fed issues upstream and the issues have not been resolved. And basic functionality in that environment has always been a little broken: virtualenv support is broken which requires user add the `--no-build-isolation` flag; and their Python configuration for finding libpython was also always a little broken. Despite this, CI works fine, which is a false flag.

## Actions
With #2634 coming in, we can recommend on Windows one of two paths:
1. anaconda or python.org installer + Visual Studio as a compiler + python test runner
2. WSL

- [ ] Change the installation instructions in the documentation
- [ ] Add existing installation instructions for these environments to the Wiki for posterity and mark them as "UNSUPPORTED".
- [ ] Add a warning to the documentation we are dropping support.
- [ ] Add a warning to the build script that GCC compilations on Windows are not officially supported.
- [ ] Close any new or outstanding issues or PRs for msys2 or mingw environments.

ktbarrett, 2022-02-18:
The mingw compiler does "just work" in anaconda. The only issue is that you can't select the mingw compiler if visual studio is available, but ultimately that shouldn't matter. There is a path forward for mingw environments and those instructions are already in the "tier 2 installation instructions" section. I don't think there is really anything to be done here.

We should start recommending WSL instead of mingw on Windows. Perhaps a blurb in the documentation to that end?

imphil, 2022-05-07:
mingw-env tests are removed from CI in https://github.com/cocotb/cocotb/pull/2964.

umarcor, 2022-05-12:
VSCode + `Remote - Containers` + `gcr.io/hdl-containers/sim/osvb` is also an interesting alternative on Windows (equivalent to the Gitpod badge).

themperek, 2024-02-10:
It seems to be working (for now) https://github.com/cocotb/cocotb/pull/3706