cocotb/cocotb #2844: Drop Python 3.6 support

ktbarrett, 2022-01-28:
Python 3.6 is EOL as of the end of 2021. 1.7 should be the last release to support Python 3.6. It might be a good idea to evaluate what can be changed by dropping 3.6. One thing I know about is OrderedDict in the compat layer (still needs to stay in the scheduler).

cmarqu, 2022-01-28:
We also might run https://github.com/asottile/pyupgrade and see what pops out.

ktbarrett, 2022-02-26:
I think we decided to do this in 1.7 instead.

imphil, 2022-02-28:
Python 3.6 is still widely used and supported by Ubuntu 18.04 and RHEL, which are both widely used in the EDA world. Unless we have strong technical reasons to remove Python 3.6 support, I am strongly against removing support for it.

ktbarrett, 2022-02-28:
Supporting 3.7+ allows us to
* drop the _py_compat library as everything in there is for compat with < 3.7
* allows the use of `from __future__ import annotations` to allow us to use newer typing functionality
* allows us to use module `__getattr__` to issue deprecations when renaming or moving individual elements of a module around

Was there anything else you thought of @cmarqu?

I was on the fence about removing it in this version or the next seeing as it EoL'd during the current development cycle (Dec 2021). But AFAIK we haven't decided to bind ourselves to OS support. Users can continue to use old versions of cocotb, they don't have to update.

cmarqu, 2022-02-28:
No, nothing more from my side. 3.7 wasn't as big a step for QoL as 3.6 was with f-strings.

ktbarrett, 2024-12-22:
Dupe of #3730.

closed 2024-12-22 by ktbarrett