cocotb/cocotb #2847: How to include Directory for Xcelium

HarshaReddy7569, 2022-01-31:
I am trying to use this in my Project, where Verilog code as some include files, while running Xcelium we will mention the directories as +incdir+<Path>.

But How to provide the directory path for compilation Here?

themperek, 2022-01-31:
You should be able set this with `EXTRA_ARGS` in your Makefile https://docs.cocotb.org/en/stable/building.html#var-EXTRA_ARGS

HarshaReddy7569, 2022-01-31:
I am providing the complete directory path like below. But still compiler is not detecting the files from the blow directory

`EXTRA_ARGS += +incdir+/rtl/pif/inc`

themperek, 2022-01-31:
You should be able to see the command that is executed in the log/terminal. If the command is right this has to be something else.

HarshaReddy7569, 2022-01-31:
Thanks.

now it is working. using the EXTRA_ARGS in Some other place.

closed 2022-01-31 by HarshaReddy7569