cocotb/cocotb #2848: Add Questa-VHPI support to Python test runner

ktbarrett, 2022-01-31:


themperek, 2022-01-31:
How we should decide if to use FLI/VHPI?

ktbarrett, 2022-01-31:
I think we can add another "language" to the toplevel. "vhdl" chooses the default, "fli" or "vhpi" override the default. The default can be "fli" for Questa for now, and when the VHPI is good we should switch.

"toplevel_lang" is really only important for how cocotb is loaded and for turning off tests that have problems in that interface. So "toplevel_lang" should be "toplevel_if" IMO.

themperek, 2022-02-01:
I am for renaming `toplevel_lang` to `toplevel_if` this than cold take `verilog`, `vhdl`, `vpi`, `fli` or `vhpi`.
Would this be ok?

ktbarrett, 2022-02-01:
Yeah that would work great. We also need to deal with the `TOPLEVEL_LANG` environment variable.

cmarqu, 2022-02-01:
Chance to introduce a new `COCOTB_TESTBENCH_IF` in addition to the old `TOPLEVEL_LANG` (with a deprecation message).

themperek, 2022-02-22:
?

1. `COCOTB_TESTBENCH_IF`
2. `TESTBENCH_IF`
3. `TOPLEVEL_IF`

ktbarrett, 2022-02-22:
I like COCOTB_TOPLEVEL_IF. One thing to consider is that cocotb.LANGUAGE is populated using `TOPLEVEL_LANG` right now. If we add support for values like `fli`, `vhpi`, and `vpi`, we are going to need to either map those back to the correct language.

ktbarrett, 2025-02-21:
Closed by #3103.