cocotb/cocotb #2851: Calling async Coroutine from  Synchronous code

HarshaReddy7569, 2022-02-02:
Hi,

I already have a large set of codes that create the Register Map and I can work with that in Software Mode, without connecting to simulation it will create a local memory like a RAL in UVM.

Now I created the Complete Python TB with CoCoTb.  I have drivers to drive the Register address and data to DUT. 

In My code, it is like Software mode and Hardware, If I am in Software mode The register Write will be stored in Python local Register Map, in hardware Mode, the Register transaction will drive to DUT using async driver function.

This async driver function needs to trigger from the synchronous code. Tried multiple ways with asyncio but I was unable to get it, Is there any way to do this?

marlonjames, 2022-02-02:
In your Software Mode, are you just running a Python script like `python script.py`?

Cocotb runs within the context of a simulator, and is at the mercy of that simulator as far as when it gets to run. Every bit of cocotb code that runs and executes your tests is only running during a callback from the simulator. There is no way to interrupt the simulation except through registered callbacks, which are created with cocotb [triggers](https://docs.cocotb.org/en/stable/triggers.html).

HarshaReddy7569, 2022-02-02:
actually, the software mode code is called as a function from the `@cocotb.test`  there it will check the Configuration is in software or Hardware mode make the decision based on that.

In Hardware Mode, it will call the async driver function from synchronous code

marlonjames, 2022-02-02:
Look at `@cocotb.function` decorator: https://docs.cocotb.org/en/stable/library_reference.html?highlight=function#cocotb.function

HarshaReddy7569, 2022-02-02:
I am using the asyncio in the sync code and calling the async driver function

```python
   loop = asyncio.get_event_loop()
   value = loop.run_until_complete(txn(cfg.dut, WRn=0, addr=self._addr)) # txn is async driver function   
```

While running above, getting the below error

```RuntimeError: Task got bad yield: <Timer of 2500000.00ps at 0x7f391d1b38c0>```

marlonjames, 2022-02-02:
cocotb is not compatible with `asyncio` in that way.

HarshaReddy7569, 2022-02-02:
okay, Thanks.

I will look into the above link.

marlonjames, 2022-02-03:
I would look into making your function async. In the "Software" mode you could just call whatever function you need and return, but in the other case you can `await coro`.

HarshaReddy7569, 2022-02-04:
Yes, That is what I am also thinking. I will rewrite the  Functions to be in async.

Thanks for your Help in Clarifying this :-)

closed 2022-02-04 by HarshaReddy7569