cocotb/cocotb #2852: Differentiate between Questa and ModelSim and use `qrun` for Questa

raysalemi, 2022-02-04:
The 3-step flow associated with Questa

* `vlog`/`vcom`
* `vopt`
* `vsim`

Is no longer recommended for Questa.  We recommend using `qrun` which is like Cadence's `irun` in that you run everything from a single command line.  `qrun` does the following:

* Figures out the file compilation order
* Implements make-like incremental compile
* Uses the correct optimization techniques for the chosen waveform tool
* Runs the simulation

**cocotb** would be better off using qrun in its simulation flow, perhaps when switching to the new run method.  This issue is to track this idea.

ktbarrett, 2022-02-04:
In what version of Questa was this introduced? We do want to support a "reasonable" set of older versions of tools. Personally I'd like to see cocotb continue to support 10.7c.

raysalemi, 2022-02-04:
It's been in place since 2019.4.

raysalemi, 2022-02-04:
10.7 was the last Questa to have  10.x version number. So there is a gap.  I suppose one could check the version number.

themperek, 2022-02-04:
Can add something like `SIM=questa_qrun`. With python runner we could "easily" check version and choose.

raysalemi, 2022-02-04:
I think it makes sense to incorporate this into the python runner rather than modify the existing system.

raysalemi, 2022-02-04:
I just wanted to get this issue recorded for the python runner.

raysalemi, 2022-02-12:
> In what version of Questa was this introduced? We do want to support a "reasonable" set of older versions of tools. Personally I'd like to see cocotb continue to support 10.7c.

Siemens EDA no longer supports 10.7. I recommend moving onto the latest release.

raysalemi, 2022-02-12:
> Can add something like `SIM=questa_qrun`. With python runner we could "easily" check version and choose.

Since Siemens EDA no longer supports 10.7, I recommend using `SIM=QUESTA`

raysalemi, 2022-02-12:
I could do the work on this if it would help.

raysalemi, 2022-02-12:
That's great.  The qrun command is much more efficient than the three-step flow.  It automatically handles ordering and incremental compilation. It also works with the Visualizer GUI (-visualizer).

Please let me know if you need technical help.

vishnutskumar, 2024-07-23:
Hey, would love to know if this is working?

ktbarrett, 2024-07-23:
Currently, no. https://github.com/cocotb/cocotb/pull/3306