cocotb/cocotb #2854: Add/Expose setup and teardown steps

suzizecat, 2022-02-08:
Hi,

Coming from a (basic) usage of (`unittest`)[https://docs.python.org/3/library/unittest.html], I think it would be nice if cocotb had some kind of  `setUp()`/`tearDown()` handles available (as described (here)[https://docs.python.org/3/library/unittest.html#unittest.TestCase.setUp])

The target would be to provide a way for the user to add some code just before/after each tests (when it is needed to provide the same setup sequence, or to ensure that the design is set back to a known status after a test.)  

Another couple of setup/teardown function could also be provided for the whole testsuite, being executed once before any test and once after the last test of the testsuite.
This could be used to setup checkers using coroutines/threads and therefore avoiding:
 - Polluting the beggining of every test
 - Re-spawning coroutines/threads.

The first situation is probably relatively easy to implement, by adding something in the `test` decorator. The resulting code looking like :
```python
if setup_function is not None : 
    setup_function(dut)
test_function(dut)
if teardown_function is not None :
    teardown_function(dut)
```

The second one might be a bit trickier, but working with the same principle. Something like (considering the full cocotb initialisation):

```python
cocotb_init() # Iinitialize the simulator link, so we can read the design and are ready to go
if setup_function is not None : 
    setup_function(dut)
run_all_tests(dut)
if teardown_function is not None :
    teardown_function(dut)
```

From a user point of view, those functions should be visible on a global scope, to avoid using them multiple times. Something like:

```python
def my_function(dut):
    cocotb.log.debug("Hey there ! Setup is ongoing !")
    cocotb.fork(my_checker(dut))

cocotb.set_test_setup(my_function)

@cocotb.test()
def test_my_unit_test(dut)
     # virtually call my_function(dut)
     # ...
```

Names could be:
 -  `set_test_setup()`
 -  `set_test_teardown()`
 -  `set_global_setup()`
 -  `set_global_teardown()`

Such setup/teardown could also be implemented on a TestFactory level, with the same kind of implementation.

marlonjames, 2022-02-08:
I have done the first part with an alternate `@test` decorator.

ktbarrett, 2022-02-08:
Is there any reason to not simply copy unittest's interface? Not sure I understand the point of the hook registering functions.

jonpovey, 2022-02-09:
I would also be interested in the ability to have a group of tests run under a shared setup/teardown, and some way to keep objects and running coroutines across those tests.
not familiar with unittest, maybe worth looking at how pytest does this too. Related to cocotb-test / upcoming integrated python runner, perhaps?

ktbarrett, 2022-02-09:
Pytest accepts unittest's TestCases. https://docs.pytest.org/en/6.2.x/unittest.html

The python runner is already merged (though it does need work). It only concerns running simulations and not dealing with with cocotb tests. It's just there to replace the makefiles.

themperek, 2022-02-09:
Do not know if this helps anyone but with pytest and python runner one can do something like this:

```python3

@pytest.mark.parametrize("testcase", ["begin, test1, end", "begin, test2, end"])
def test_my_dut(testcase):

    runner = get_runner()()
    runner.test(
       ...
      testcase = testcase
    )
```

Can use `testcase`,`py_module` or anything else as parameters.
This can run in parallel if wanted.

suzizecat, 2022-02-09:
> Is there any reason to not simply copy unittest's interface? Not sure I understand the point of the hook registering functions.

On my side, I would gladly have the unittest interface which seems a bit more powerful than the current cocotb one.
I am not sure how it would interface with cocotb internal, but I'm a beginner on the topic of unit testing anyway.

jonpovey, 2022-02-10:
> So what about this for TestCases?

Keen for something approximately like this, haven't spent time looking at good structuring for how these things get written.
However... please all lowercase `setup()` and `teardown()`!

suzizecat, 2022-09-28:
Had seemingly never saw the notification, but @ktbarrett proposition looks good to me.
You might want to have the tests as proper member functions (with `self`) in order to easily share resources.

Nested testcases (with individual fast fails) would be great ! 
In example, having a root test-case with fast fail true, but the "leaf" test-cases without the fast fail would let the test-cases run until completion but stop the root test suite upon failure.