cocotb/cocotb #2869: Odd behavior when performing += in simulation

wh14h, 2022-02-20:
I have a simulation that initializes a size 8 std_logic_vector to 0 and then adds 1 for simulation purposes, like

> 
    dut.wr_data.value = 0
    ... wait on timer ...
    ... for loop
    dut.wr_data.value += 1
    ... end for loop

And wr_data jumps from U to 0x80 instead of 0 and increments by 1, like 0x80, 0x81, 0x82... which is not desired.

Doing ``dut.wr_data.value = dut.wr_data.value + 1`` instead of using +=1 does what is expected, initializing to 0 and incrementing by 1 with each for loop, like 0, 1, 2...

I looked closer at ``binary.py``, where the ``__iadd__`` function for += is implemented and this might be an issue with ``_convert_to_unsigned`` as it simply performs ``bin(x)``, which is then passed to ``_adjust_unsigned`` where length-dependent operations occur, namely turning 1 into 10^7 because bin(1) == "0b1" and "1" is then passed to ``_adjust_unsigned`` when ``_adjust_unsigned`` is expecting "00000001" as input to return big endian 1 in an 8-bit storage space.

I've create a demo to showcase this issue and can send over the vhdl file, the python sim file, and the Makefile as needed.

======================================

- cocotb version: 1.6.2 from pip
- Operating system: WSL2 Ubuntu18.04 kernel 5.10.16.3-microsoft-standard-WSL2 (64-bit)
- Simulator: GHDL 2.0.0-dev
- Python version: Python 3.6.9 from system. Using same version in a virtual environment.

ktbarrett, 2022-02-23:
BinaryValue is very broken. I recommend to not use any of the features besides:
* `.integer`
* `.signed_integer`
* `.binstr`

It will be totally replaced in the future.

closed 2022-02-23 by ktbarrett

marruffo1024, 2022-04-01:
Hello! Is there any estimate on how long it will take? How is the progress so far? I am just inquisitive about it, for as I am also facing obstacles while using it, and wondering about what to do in the meantime (introduce workarounds?). Nevertheless I appreciate the work it being done here! The library has been a powerful and very useful tool to me.

Also, on that note, is there anything I can do to help regarding the development of that feature or in general?