cocotb/cocotb #2870: Installation trouble - Failed building wheel for cocotb

AlbertHoevenaars, 2022-02-21:
After an attempt to use cocotb a few months ago without success, I want to give it a second shot. I am using Windows 10 as OS.
Now I want to install the tools on another computer, but the installation gives me some errors. For me the installation description is rather poor. Sometime its unknown where to download tools and what versions are needed. 
I manged to install Python (version 3.10), that's rather easy. The same counts for Conda. But the make file installation gives me trouble, but that can be skipped, is only needed to run the testbenches. But I still don't know where to download and which version. Installation of Microsoft Visual C++ is much more problematic. Only Visual C++ or also Studio? its should be version 14+, I installed version 16. 
But the installation of cocotb gives me really errors.
`
      "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\bin\HostX86\x64\cl.exe" /c /nologo /O2 /GL /DNDEBUG /MD /W4 -DGPILOG_EXPORTS= -D__STDC_FORMAT_MACROS= -DWIN32= -Icocotb\share\include -IC:\Users\Albert\AppData\Local\Programs\Python\Python310\include -IC:\Users\Albert\AppData\Local\Programs\Python\Python310\Include "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" /EHsc /Tpcocotb\share\lib\gpi_log\gpi_logging.cpp /Fobuild\temp.win-amd64-3.10\Release\cocotb\libs\libgpilog\cocotb\share\lib\gpi_log\gpi_logging.obj /permissive-
      gpi_logging.cpp
      C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\include\cstdio(12): fatal error C1083: Cannot open include file: 'stdio.h': No such file or directory
      error: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.31.31103\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb, which is required to install pyproject.toml-based projects
`
I did an update for pip, which was proposes with an earlier error message.
Don't know what is going on here. Placed the installation I did in a document. Is attached. Especially the red lines in the doc are unclear for me. Sorry I am a hardware guy, with only little software experience.
[CoCotb.docx](https://github.com/cocotb/cocotb/files/8108204/CoCotb.docx)
e

ktbarrett, 2022-02-21:
> I manged to install Python (version 3.10), that's rather easy. The same counts for Conda.

We only support anaconda/miniconda installations on Windows at this time. I would actually remove the system Python installation otherwise it may be confused with conda.

> But the make file installation gives me trouble

Per [the installation documentation](https://docs.cocotb.org/en/stable/install.html#installation-of-prerequisites) you can run `conda install -c msys2 m2-base m2-make` within a conda shell to install make and a C/C++ compiler into that environment. This is the only supported way to install make.

> Installation of Microsoft Visual C++ is much more problematic

If installation of MSVC is problematic, the above `conda install` command installs the GCC C/C++ compiler, so you don't need MSVC. If GCC is acceptable I would remove your MSVC installation as setuptools will always attempt to build with MSVC if it is present.

> Only Visual C++ or also Studio?

Only the Visual C++ compiler, Visual Studio is an IDE, but only if you plan on using Microsoft's compiler over GCC.

> But the installation of cocotb gives me really errors.

Can you build any software at all with your MSVC installation? That error is indicative of misconfiguration of MSVC or your system as a whole. You can see if any of [these solutions](https://stackoverflow.com/questions/44219901/cannot-open-include-file-stdio-h-visual-studio-community-2017-c-error) help you. If you want to stick with MSVC.

We will be providing pre-built wheels sometime soon (tm) for all simulators on Windows besides Questa's VHDL support (since the FLI headers are not permissively licensed). If you continue having issue it might be best to hold off until we can provide the wheels. We can't remotely debug user's environments.

AlbertHoevenaars, 2022-02-21:
ktbarrett, thanks for your reply. 
I removed all the software that I installed, and deleted the install directory of Python, to get a clean start. Then I did the following:
`
•	Install miniconda3
•	Open Anaconda prompt
•	Give command: 
		conda install -c msys2 m2-base m2-make
•	Message:
EnvironmentNotWritableError: The current user does not have write permissions to the target environment.
  environment location: C:\ProgramData\Miniconda3
•	Install Microsoft Visual C++
•	Command (still in Anaconda prompt window): 
pip install cocotb
Error message I get is:
creating build\lib.win-amd64-3.9\cocotb\share\lib\verilator
  copying cocotb\share\lib\verilator\verilator.cpp -> build\lib.win-amd64-3.9\cocotb\share\lib\verilator
  running build_ext
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly
`
Any suggestions?

ktbarrett, 2022-02-21:
> EnvironmentNotWritableError: The current user does not have write permissions to the target environment.

If you installed your miniconda with admin privileges, you need an elevated prompt to install any packages. You will need to start anaconda prompt by right clicking and "run as admin" to get the elevated prompt to be able to install packages. Instead of doing that I would recommend you install miniconda into your user directory, i.e. do not check "install for all users". 

> error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/

Did you restart after installing? A fresh login is required for MSVC directories to be added to the PATH for a command prompt to pick up.

AlbertHoevenaars, 2022-02-21:
ktbarrett, I did not restart the computer.
After restarting I opened the miniconda promt as admin. Give the command "pip install cocotb" but getting the same error.
I followed your link for visual C++, but that was the same as I used. Downloaded from the site:vs_buildtools__d66a8409342a4f1d8cab2ecb7f8d0edc.exe". Was indeed the samefile I already donwloaded and installed.

Error message:
  creating build\lib.win-amd64-3.9\cocotb\share\lib\verilator
  copying cocotb\share\lib\verilator\verilator.cpp -> build\lib.win-amd64-3.9\cocotb\share\lib\verilator
  running build_ext
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly

Any other suggestion?

ktbarrett, 2022-02-21:
Did you run `conda install -c msys2 m2-base m2-make` in an elevated conda prompt?

AlbertHoevenaars, 2022-02-21:
I uninstalled miniconda and installed it again, but only for me and not for all users. 
On the command conda install -c msys2 m2-base m2-make I get the message "C:\Users\Albert\AppData\Local\Temp\conda-1691\conda.tmp - Access is denied."

After command pip install cocotb I get the message (sevarel times) " PermissionError: [WinError 5] Access is denied: 'C:\\Users\\Albert\\AppData\\Local\\Temp\\pip-req-tracker-8imsj0dk\\c9deb16598cb7c7ff74899d960af0bf229d29c84842508eaee111c3a'

Messages on different files in the Temp directory.
Maybe this helps figuring out what the problem is.

ktbarrett, 2022-02-21:
You should have permission to write to your own Temp directory. Something is very wrong.

ktbarrett, 2022-02-23:
That's great to hear! I will take those instructions and add them to a wiki page.

AlbertHoevenaars, 2022-02-25:
Now the installation seems to be OK, I would like to run the examples. But I cannot find them. I searched C:\ProgramData\Miniconda3\Lib\site-packages\cocotb but they are not there. Where can I find the examples?

ktbarrett, 2022-02-25:
The examples are not distributed with the package, you will have to clone or download a copy of the repo.

ktbarrett, 2022-03-04:
Have your questions been answered?

AlbertHoevenaars, 2022-03-04:
Oh yes. Thanks for the help!

closed 2022-03-04 by ktbarrett