cocotb/cocotb #2887: Make it clearer that environment variables must be exported in make

ktbarrett, 2022-03-02:
From https://github.com/cocotb/cocotb/discussions/2885#discussioncomment-2278102. It seems that some environment variables are exported (like MODULE and TOPLEVEL_LANG) and other are not, which makes it even more confusing.

aaravsinghbisen, 2026-11-04:
Hi! I'd like to work on this. I checked on Icarus 12 that only the variables in `COCOTB_ENV` (plus a few exported in `Makefile.inc`) reach the simulator, while others like `COCOTB_ENABLE_PROFILING` don't unless set in the shell, on the `make` command line, or exported. I have a docs change ready that documents this. Could you assign this issue to me?

ktbarrett, 2026-11-05:
We do not need to list which ones are export or not, it's better to just mention that they should all be exported if users want them to be reliably used.