cocotb/cocotb #2899: Modifying SIM_ARGS in the Makefile does not update runsim.do

JC-FPGA, 2022-03-08:
I'm using cocotb v1.6.2 on Ubuntu 20.04.1

When updating SIM_ARGS in the Makefile, the automatically generated sim_build/runsim.do does not get updated.  For example, changing
SIM_ARGS +=-gIMAGE_WIDTH=1280
to
SIM_ARGS +=-gIMAGE_WIDTH=1920

the vsim command stays at the old value:
`vsim -gIMAGE_WIDTH=1280 -onfinish exit -foreign "cocotb_init  ....`

After changing SIM_ARGS, a workaround is running "make clean" or simply deleting the runsim.do file.  Since sim_build/runsim.do is an "autogenerated" file, I would expect it to be updated upon a change to Makefile.

themperek, 2022-03-09:
> I would expect it to be updated upon a change to Makefile.
This is nothing cocotb can do. It is how makefiles work. 

You can add your makefile to `CUSTOM_SIM_DEPS`.

themperek, 2022-03-09:
Acuity one could add this inside `cocotb`.

ktbarrett, 2022-03-09:
When projects become split across multiple makefiles that becomes impossible to decide. Tools like CMake handle this by creating a cache file of the variables and seeing if there is a diff between the current variables and the old variables.

closed 2022-04-16 by github-actions[bot]