cocotb/cocotb #2910: How to run regressions (usage of TestFactory)

PiyushSaini09, 2022-03-11:
cocotb version --> v1.6.0.
Linux Distro --> RedHat
Python --> v3.9.0

===========================================================================

Hi-
I've a bunch of tests written in my file. I want to run regression on those which basically means that I want each test to run one after the other with each test running from time = 0. 
I referred to the issue #2874 and tried ktbarrett's solution to do the same.

> # Makefile
> 
> test1:
>     $(MAKE) sim MODULE=tests TESTCASE=test1
> 
> test2:
>     $(MAKE) sim MODULE=tests TESTCASE=test2 
> 
> regression: test1 test2
> 
> include $(cocotb-config --makefiles)/Makefile.sim

However, in this case only the first test runs and the simulator stops. Second test never runs.
In order to run this I typed `make` in my console. What am I doing wrong in this case?

I also reffered to `TestFactory` but I couldn't understood how to use it or what's the purpose. How do tests get generated from it? Also, the use of add_option is not very clear from the documentation.

cmarqu, 2022-03-11:
Calling `make` without any arguments just executes the first target.
Move your `regression` target up to the top and you should be fine.

PiyushSaini09, 2022-03-11:
Understood.
Thanks a lot.
I'll close the issue now.

closed 2022-03-11 by PiyushSaini09