cocotb/cocotb #2914: Running cocotb in Windows

PiyushSaini09, 2022-03-15:
Hi-

I've been using cocotb on linux for a while now but due to some reason I had to move to windows 10.
I've installed modelsim as my simulator and was trying to get cocotb running in that.
Been using VSCode for the writing the code (system verilog as well as python TB).

The issue I've been facing is I dont know how to run the simulation. VScode has integrated terminals so I tried using the "cmd" of that. Whenver I run the code by `make` the following error is thrown:
[image]

I dug through the `makefile.sim` and found out the issues in these:
[image]

Are these commands windows compatible?

If I manage to solve all this will cocotb automatically recognize modelsim (installed in my system) as the simulator and run my simulation?

For reference I'm also attaching my environment variable (because windows makes it difficult to configure everything):
[image]

Thank you.

=====================================================================================
Version - cocotb v1.6.2
OS - Windows 10 64-bit
Python - 3.10.2

themperek, 2022-03-15:
Seems you are missing some Unix tools. Did you follow install intructions https://docs.cocotb.org/en/stable/install.html#installation ?

PiyushSaini09, 2022-03-15:
Yes. I followed all the instructions. 
Can you please clarify the steps to run the simulation in windows as well?
I thought running from miniconda command prompt or windows command prompt would work.
But it did not work for me

themperek, 2022-03-15:
miniconda command prompt  sets some paths that are necessary
You can try to activate conda via `conda init`. It seems to work in power shell.

You can alternatively try to use https://github.com/themperek/cocotb-test  (similar will be soon part of cocotb #2634 )

PiyushSaini09, 2022-03-15:
When I try to invoke conda from `cmd` I encounter the following error:
[image]

Does this mean that conda was not properly installed?

themperek, 2022-03-15:
You have to add conda to %PATH%

PiyushSaini09, 2022-03-15:
Did `conda init` and got the following log:
[image]

But am facing the same issue as before.
[image]

Do the forward and backward slashes(in the path of the Makefile.sim) make any difference?

marlonjames, 2022-03-15:
Did you install cocotb from the conda command prompt?
If so, you need to open a conda command prompt to be in the correct environment.

ktbarrett, 2022-03-15:
If you are running Windows 10 I recommend you use a virtual machine or WSL to run Linux and not bother with alternative Unix subsystems for Windows. They are all painful and unintuitive to use. We are updating our recommendations for the 1.7 release.

PiyushSaini09, 2022-03-16:
@marlonjames I tried running from conda prompt and got this:
[image]

> If you are running Windows 10 I recommend you use a virtual machine or WSL to run Linux and not bother with alternative Unix subsystems for Windows. They are all painful and unintuitive to use. We are updating our recommendations for the 1.7 release.

I'm thinking to shift to VM. It will be easier to have a complete linux system.

closed 2022-04-23 by github-actions[bot]