cocotb/cocotb #2923: make cannot find mti.h

dalleggio, 2022-03-22:
* the cocotb version used: 1.6.2
* the operating system and version (32/64 bit): Kubuntu 20.04.4 LTS, 64-bit
* the simulator and version (32/64 bit): Modelsim 2020.4, 32-bit
* the Python version: Miniconda (4.11.0), Python 3.7.1
* (part of) the log file (cleaned of proprietary information).
(py3_32) $ make GUI=1 WAVES=1 all 
...
make -f Makefile results.xml
make[1]: Entering directory '/home/anthony/.../tb/sim/_proj'
ERROR: cocotb was not installed with an FLI library, as the mti.h header could not be located.
If you installed an FLI-capable simulator after cocotb, you will need to reinstall cocotb.
Please check the cocotb documentation on ModelSim support.
make[1]: *** [/home/anthony/miniconda3/envs/py3_32/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:75: /home/anthony/miniconda3/envs/py3_32/lib/python3.7/site-packages/cocotb/libs/libcocotbfli_modelsim.so] Error 1
make[1]: Leaving directory '/home/anthony/git_repos/Q_Enyx_530_anthony/jhwplus-hw/tb/sim/_proj'
make: *** [/home/anthony/miniconda3/envs/py3_32/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

Parts of Makefile

SIM = modelsim
TOPLEVEL_LANG = vhdl
COCOTB_HDL_TIMEUNIT = 100ps
COCOTB_HDL_TIMEPRECISION = 100ps
TOPLEVEL = jhwplus_top
MODULE = test_0
SIM_ARGS += work.glbl
SCRIPT_FILE = wave.do

ProjectDir = $(shell pwd)
BranchDir = $(ProjectDir)/../../..
SourceDirectoryPath = $(BranchDir)
DestinationDirectoryPath = $(BranchDir)

all: clean_all run_prep

VERILOG_SOURCES=...
VHDL_SOURCES=...

include $(shell cocotb-config --makefiles)/Makefile.sim

run_prep:
        perl ../../scripts/update_sim_src_files.pl $(SourceDirectoryPath) $(DestinationDirectoryPath)

clean_all:
        @rm -rf sim_build

imphil, 2022-03-22:
The error is here:

```
ERROR: cocotb was not installed with an FLI library, as the mti.h header could not be located.
If you installed an FLI-capable simulator after cocotb, you will need to reinstall cocotb.
```

Please refer to the documentation at https://docs.cocotb.org/en/stable/simulator_support.html#mentor-siemens-eda-modelsim to resolve this issue.

dalleggio, 2022-03-22:
I guess I should have said that for the past two days I read and googled that error message multiple times and made sure I followed all the steps.  I reinstalled cocotb 3 times AFTER modelsim and made sure that the PATH variable includes the modelsim bin directory that contains vsim and vdbg.  Here's some proof

(py3_32) $ which vsim
/opt/Modelsim/modelsim_dlx/bin/vsim
(py3_32) $ ls -l $(dirname $(which vsim))/../include/ | grep mti
-r--r--r-- 1 root root  33074 Oct 13  2020 mti.h
-r--r--r-- 1 root root  13711 Oct 13  2020 mtiq.h
-r--r--r-- 1 root root   2978 Oct 13  2020 mti_traverse.h
-r-xr-xr-x 1 root root   1557 Oct 13  2020 mti_vpi_user.h

Thanks
Anthony

imphil, 2022-03-22:
How did you reinstall cocotb? It seems likely that the reinstallation used a cached version of cocotb and didn't rebuild it.

Try with `pip3 uninstall cocotb; pip3 install --verbose --no-cache-dir cocotb` and please paste the output if it still doesn't work.

dalleggio, 2022-03-22:
That was the issue - reinstalling from the cached version.  
I suggest adding that detail in the docs.
Thanks Philipp!
Anthony

closed 2022-03-22 by dalleggio

imphil, 2022-03-23:
Great, thanks for letting us know! 

There are developments which will avoid using having to rely on the `mti.h` header provided by the user, which should be even better than documentation :-)