cocotb/cocotb #2929: Properly rearming a callback so that it does not get removed

ucfssl, 2022-03-24:
Greetings,

I am attempting to use `cocotb.simulator.register_value_change_callback()` to monitor a signal as it changes. Consider the following minimal [non-]working example:

```
module foo(input a, b, output c);
    wire d;
    assign d = a | b;
    assign c = d ^ 1'b1;
endmodule
```

```
import cocotb
from cocotb.simulator import register_value_callback
from cocotb.triggers import Timer

def callback(dut):
    print("changed")

@cocotb.test()
async def test_fixture(dut):
    register_value_callback(dut.d._handle, callback, 3, dut)

    for i in range(16):
        dut.a = i & 1
        dut.b = (i >> 1) & 1
        await Timer(10)
```

I would expect that this callback remains registered through the lifetime of the simulation, as per VPI would dictate. However, the code in https://github.com/cocotb/cocotb/blob/1536fd14dddaed797ad5d68858baad1a5f578076/cocotb/share/lib/vpi/VpiImpl.cpp#L645-L654 effectively unregisters the callback unless it is set back to the `GPI_PRIMED` state. There is no clear way to do this from the Python side. Now, we could try to register the callback on every iteration of the simulation by doing:

```
    for i in range(16):
        # ...
        await Timer(10)
        register_value_callback(dut.d._handle, callback, 3, dut)
```
Due to https://github.com/cocotb/cocotb/issues/732 this leaks memory all over the place. For very long simulations this causes the simulator to eventually crash [which is the issue I am running into and what motivates this report]. I tried to deregister the callback:
```
@cocotb.test()
async def test_fixture(dut):

    for i in range(16):
        cb  = register_value_callback(dut.d._handle, callback, 3, dut)
        dut.a = i & 1
        dut.b = (i >> 1) & 1
        await Timer(10)
        cb.deregister()
```

However, this approach still leaks memory which is what I am trying to avoid.

Other things I tried:
- Removing the check on the launcher for the callback to skip a cleanup. This breaks other stuff in cocotb, even if we allow the callback to go back into the `GPI_PRIMED` state.
- I wrote the attached patch to try and add a way to rearm the callback from Python. It does not work for some reason. I still seem to be corrupting something somewhere and of course the callback does not go to the `GPI_PRIMED` state.
```
from types import SimpleNamespace

import cocotb
from cocotb.simulator import register_value_change_callback
from cocotb.triggers import Timer

def callback(cb_data):
    cb_data.log.info(f"callback: {cb_data.msg}")
    cb_data.cb.rearm()

@cocotb.test()
async def test_fixture(dut):
    cb_data = SimpleNamespace(msg='called', log=dut._log)
    cb = register_value_change_callback(dut.d._handle, callback, 3, cb_data)
    cb_data.cb = cb

    for i in range(16):
        dut.a.value = i & 1
        dut.b.value = (i >> 1) & 1
        await Timer(10)                                                                                                                                                                                                                                       
```

On output, I get:
```
[...]
     0.00ns INFO     Found test test.test_fixture
     0.00ns INFO     running test_fixture (1/1)
     0.00ns CRITICAL VPI: Callback state 2
     0.00ns INFO     callback: called
     0.00ns CRITICAL VPI: Callback state 1
     0.00ns CRITICAL VPI: Callback state 0
Userdata corrupted!
     0.00ns CRITICAL VPI: Callback state 2
     0.01ns CRITICAL VPI: Callback state 0
     0.01ns CRITICAL VPI: Callback state 0
     0.02ns CRITICAL VPI: Callback state 0
     0.02ns CRITICAL VPI: Callback state 0
[...]
```

I guess my questions here are:
- Why is it necessary to unregister these callbacks after they are executed?
- How can I make them permanent?
- Is there a more proper way to do what I want to do?
- What did I miss on my patch?

Thank you for your time.

Cheers,
Orlando.

[rearm_callback.zip](https://github.com/cocotb/cocotb/files/8344333/rearm_callback.zip)

marlonjames, 2022-03-24:
You should use the `Edge()` trigger: https://docs.cocotb.org/en/stable/triggers.html#cocotb.triggers.Edge
There are limitations to the GPI libraries with `gpi_register_value_change_callback()`. It does not support multiple callbacks for a signal handle, and so you should restrict yourself to the Simulator triggers in Python code. (https://github.com/cocotb/cocotb/issues/2208 is an issue to address that shortcoming, but it targeted at GPI extensions)

ktbarrett, 2022-03-24:
> as per VPI would dictate

Where did you read that?

marlonjames, 2022-03-25:
Maybe he is talking about how `cbValueChange` callbacks are repetitive callbacks, not one-time callbacks like simulation time callbacks.
It will remain registered unless you remove it.
GPI will remove it if the external system doesn't need to know about value changes any longer.

ucfssl, 2022-03-28:
> > as per VPI would dictate
> 
> Where did you read that?

Per IEEE 1800, `cbValueChange` remain active until explicitly removed with `vpi_remove_cb` as @marlonjames clarified. I should have been more specific on that, my apologies.

> You should use the Edge() trigger: https://docs.cocotb.org/en/stable/triggers.html#cocotb.triggers.Edge

The solution/workaround that @marlonjames proposed seems to works for what I want to do. Initial profiling suggests that the memory leak is not present. I am guessing that there's something on the cocotb scheduler going on behind the scenes that reactivates the callback? I would like to know how this works behind the scenes [a pointer to the relevant piece of code would suffice].

closed 2022-05-06 by github-actions[bot]