cocotb/cocotb #2930: Support for archs other than x86_64

allexoll, 2022-03-31:
Currently cocotb only supports x86_64. 

As Apple is turning away from x86_64 and into arm64, the current Mac user base that was previously running it either on macos-x86_64 or linux-x86_64 in docker/VM will need tools to work on their machine. 

The open source HDL tooling might be especially well setup for this since it seems that this is not on commercial tools's roadmap.

I commented that I was able to build successfully for mac_arm64 and linux_aarch64 in #2924 but this is a larger discussion about the supported platforms of cocotb.

One potential issue is that currently github action does not have arm runners, but it's still possible to have them trough [qemu + docker ](https://github.com/dbhi/qus). 

Happy to talk about it

ktbarrett, 2022-03-31:
> One potential issue is that currently github action does not have arm runners, but it's still possible to have them trough [qemu + docker ](https://github.com/dbhi/qus).

Looks promising. Can you open a PR? It works a bit differently than I expected since GHA supports using docker containers in "uses" blocks.

imphil, 2022-03-31:
Let's maybe take them separately.

* aarch64 Linux is getting more and more support with proprietary EDA tools as well (I've seen it in the release notes for VCS and Questa at least). So that's something where apparently users are, and quite in line with our platform support policy. Sounds like an easy one to add cocotb support for as well.
* M1 macs are a bit a different story that I'd like to learn more about. We have a tiny user base on macos anyways, and I'd guess it's even smaller on M1 macs. What's the simulator support story on M1 macs? Are there precompiled versions of Icarus, GHDL, or Verilator available? Or are there other simulators supporting Mac? I'm especially wondering about precompiled versions, since even if we don't officially make M1 macs a supported platform and build wheels for it, users can still compile cocotb from source.

allexoll, 2022-04-01:
linux aarch64 seems to be the easiest one to add, and the most usefull one since it could also be run trough docker on Macs anyway.

Regarding Macos arm64, icarus, ghdl, verilator are all available precompiled, trough the most used package manager on mac `brew`

```sh
brew install icarus-verilog
brew install ghdl
brew install verilator
```

both linux-aarch64 and macos-arm64 cocotb build succeed but i'm unsure what the framework is for testing it would be, especially since it has to integrate with GA.

So far the way to do it would be:
* x86_64 builds don't need to change
* other  archs needs to call the QUS action (to install arch runtime) but then the run needs to be inside a container of said architecture. This is needed because while qus allows you to run foreign binaries, a lot of tooling relies on `uname -m` to decide what version is being built. running the build inside a docker addresses that issue. 

There's also the issue of limited run time for GA jobs (6 hours i think?) which might be an issue for some project since QEMU is 5-10x slower than native. 

I'd be happy to have a look but i'm not familiar yet enough with cocotb's workflows and how that would tie in with @imphil 's wheel build

ktbarrett, 2025-10-17:
2.0 and future versions will support MacOS arm64.

closed 2025-10-17 by ktbarrett